bambulab/BambuStudio #7224: Bambu Studio crashes constantly

pyyanji, 2025-06-11:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Laptop: Dell Precision 5680
CPU: Intel i7-13700H
GPU: Nvidia RTX A1000 6GB Laptop

### Printer

Bambu Lab X1C

### How to reproduce

1. Open project file
2. Slice
3. Start checking different layers in preview window
4. Crash occurs immediately after slice or shortly after 

### Actual results

Bambu Studio crashes and closes.
Restricting CPU affinity to fewer cores does not help.

### Expected results

App should continue running so I can print :)

### Project file & Debug log uploads

[bambu-studio.exe.27248.zip](https://github.com/user-attachments/files/20684829/bambu-studio.exe.27248.zip)

[Dronejalka_12kpl.zip](https://github.com/user-attachments/files/20684841/Dronejalka_12kpl.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-06-11:
[image]
Hello，If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

pyyanji, 2025-06-11:
That seems to have solved the issue. :)
Thank you so much!

closed 2025-06-11 by pyyanji

Haidiye00, 2025-06-11:
Okay, thank you for your feedback. We will continue to iterate and optimize