bambulab/BambuStudio #7228: 使用3D建模进行布尔运算产生误差

imycard, 2025-06-11:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

CPU: AMD Ryzen 7 8845HS
Memory: 64 GB
Display Card: AMD Radeon 780M

### Printer

Bambu Lab A1

### How to reproduce

使用Bambu Studio进行简单的3D建模，在模型里放入了两个两个圆角矩形，在没有进行布尔运算-并集时，两个矩形的顶面距离是28mm（也是我设计的距离），进行并集后，距离变为27.990mm

[圆角矩形3mf.zip](https://github.com/user-attachments/files/20685604/3mf.zip)

### Actual results

布尔运算-并集后模型位置产生了误差
[image]
[image]

### Expected results

布尔运算-并集后模型位置保持正确

### Project file & Debug log uploads

[圆角矩形3mf.zip](https://github.com/user-attachments/files/20685636/3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-06-11:
[image]
Hello, I have checked and found that there is indeed a certain degree of error after Boolean calculation. This may be due to the operation of floating-point numbers or some decimal point selection in the operation. At present, I believe that this accuracy is sufficient

Haidiye00, 2025-06-11:
Hello, the range of Boolean error is approximately within plus or minus 0.02mm, which I think is acceptable. What do you think

imycard, 2025-06-12:
@Haidiye00 这个问题是否需要改进由你们决定，我只是发现了这个问题并提出，实际使用中的影响不大，我对boolean计算原理并不了解，只是觉得一些简单的模型搭建修改使用Bambu Studio比较简单，没有门槛。专业的建模软件里没有这个问题，当然Bambu Studio主要用于切片，不能按照专业建模来要求。

Haidiye00, 2025-06-12:
Okay, considering that the 3D printing accuracy is already met, we have decided not to make any modifications for now~

closed 2025-06-12 by Haidiye00