bambulab/BambuStudio #7233: "Please change the nozzle settings on the printer"

eHanseJoerg, 2025-06-11:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

macOS latest

### Printer

P1S

### How to reproduce

1. Change nozzle on a printer
2. In Bambu Lab, go to DEvice
3. Try to change nozzle diameter
4. Observe that nozzle diameter change is grayed out and it says "Please change nozzle diameter on the printer"

Problem: the nozzle diameter setting is buried deeply in the printer on screen menu

### Actual results

User can not find the nozzle diameter setting in the printer.

### Expected results

On a P1S with a small LCD screen, provide easier access to the nozzle diameter menu.

### Project file & Debug log uploads

[RESN_OH4A6A1-15_T_O TRIM BUTTON_-_V54.3mf.zip](https://github.com/user-attachments/files/20691379/RESN_OH4A6A1-15_T_O.TRIM.BUTTON_-_V54.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

leejunweisg, 2025-06-18:
I am experiencing this issue as well. I used to be able to change the nozzle settings directly within Bambu Studio (on windows), but now it says to change it on the printer instead.

chalupasalad, 2025-06-19:
I am also experiencing this issue. I had to go back to 2.00.03.54(last previous version installed) to be able to change it and start a print remotely.

JohnSmithCYM, 2025-06-26:
Thanks for your feedback, we have disabled nozzle setting for all printers since V2.1.0 due to nozzle change option should be confirmed near the printer.You can refer to Bambu Wiki to get the way to set nozzle settings on the printer's screen https://wiki.bambulab.com/en/p1/maintenance/complete-hot-end-assembly.

milomarv, 2025-06-30:
Not only for P1S, it is buried deeply in the menu also for X1C. 
Control Tab > Nozzle and Extruder > Nozzle Type > then I am able to change it.
At least would be cool if you would add the manual for X1C to this wiki: https://wiki.bambulab.com/en/p1/maintenance/complete-hot-end-assembly as well. Had to search a bit and found it with googling…

shuchow, 2025-07-05:
> https://wiki.bambulab.com/en/p1/maintenance/complete-hot-end-assembly

Why in the world would Bambu think this is a good idea?

popcoondev, 2025-07-06:
I’m experiencing the same issue and find that the usability has significantly deteriorated. I respectfully request that you restore the previous behavior as soon as possible.

Warhawk24747, 2025-07-10:
Yes, this is a serious regression in features. Please allow us to change the nozzle size using the slicer it is significantly easier to access.

MooMinIL, 2025-07-13:
This is messed up. 
My current setup (A1 Mini + AMS Lite) has my printer screen on the far side from me (printer oriented to the side) and so I RARELY use the printer screen for any setting changes as it requires either physically turning the printer towards me or bending over the printer to awkwardly operate the screen.
Part of the selling point for the A1 series for me was fast swap of nozzles and I've been doing that a ton.
I seriously would not have updated had I known this was a change (BTW, is it just me or is this not even mentioned in the release notes? https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-2-1-0)

> [...]we have disabled nozzle setting for all printers since V2.1.0 due to nozzle change option should be confirmed near the printer.

Please bring back the capability, even if you lock it behind an 'advanced mode' toggle or put up a warning or something. Some of us rely on this kind of functionality and taking it away is poor practice and alienates your customer base.

TheRealBenForce, 2025-08-19:
This is a bad change. I absolutely can not find this on the A1 screen.

JohnSmithCYM, 2025-08-20:
[image]

https://wiki.bambulab.com/en/a1/maintenance/replace-hotend You can refer to this to get a A1 hotend setting.

veitch123, 2025-08-21:
Argh

mageofmarkarth, 2025-08-22:
Just done the update to find this out. Talk about downgrading. It's not exactly an easy option to find either on the A1. How about listening to your customer base and putting the setting back into the slicer software so we can do it while setting up the print? Like it was before.

malventano, 2025-08-22:
Even worse, on H2D, user is unable to pre-configure / slice a project with anything other than the last-seen nozzle size:

[image]

^ Choosing either option at the prompt results in the values reverting back to 0.4, making it impossible to choose another size for slicing.

mytechguyri, 2025-12-29:
> Thanks for your feedback, we have disabled nozzle setting for all printers since V2.1.0 due to nozzle change option should be confirmed near the printer.You can refer to Bambu Wiki to get the way to [...]

Yes... .We KNOW you disabled it since v2.1.0.... but for absolutely no good reason.... why can't you leave it enabled in Bambu Studio?  Why am I forced to navigate deep into the printer menus to change a setting I used to be able to easily change in Bambu Studio?   why not just "we RECOMMEND you change it directly on the printer, but if you insist, you can change it here too'?