bambulab/BambuStudio #7238: Sidebar won't fully expand

AaronRosie24, 2025-06-11:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

8th gen core i7 intel

### Printer

bambu lab p1s

### How to reproduce

the sidebar will not full expand. half of the sidebar is cutoff. I tried closing/reopening, restarting my computer, updating bambu studio

[image]

### Actual results

the sidebar will not full expand. half of the sidebar is cutoff. I tried closing/reopening, restarting my computer, updating bambu studio
[image]

### Expected results

i want to see the full sidebar

### Project file & Debug log uploads

issue happens with any project file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-15:
https://github.com/Haidiye00/BambuStudio/actions/runs/15534911769
[image]
This is a temporary version, please try it out~thanks

1.open this zip software,come to preivew tab, then close software. find this BambuStudio.conf,and modify enable_sidebar_resizable 's value
[image]
2.reopen this  zip software, come to preivew tab,then press "ctrl+w"
[image]

then you can adjust sidebar's width

crookz, 2025-06-26:
Please release this fix ASAP - its made bambu unusable. Do NOT cut out orca slicer then sit on critical fixes like this for weeks. We cant access half the settingscor loose most of our screen realestate

crookz, 2025-06-26:
> Hello, we apologize for the inconvenience. In future versions, we will default to supporting the adjustable width of the sidebar

Please just release a patch now, dont wait weeks, or get rid of your new secruity and let us use orca slicer so at least we have an alterantive if your software is shipped broken

Haidiye00, 2025-06-27:
> [image] https://github.com/Haidiye00/BambuStudio/actions/runs/15534911769 这是临时版本，请试用一下~谢谢
> 
> 1.打开这个 zip 软件，进入 Preivew 选项卡，然后关闭软件。找到这个 BambuStudio.conf，[image]并修改 enable_sidebar_resizable 的值 2.重新打开这个 zip 软件，进入 preivew 选项卡，然后按 “ctrl+w” [image]
> 
> 然后你可以调整侧边栏的宽度

Hello, a temporary solution has been provided here

crookz, 2025-06-29:
As stated above im not ok with downloading a random build

Haidiye00, 2025-06-30:
Suggestion: After using up the temporary package, you can switch back to the official package

Nofreakinway, 2025-07-12:
I tried editing the sidebar settings to true as above, but they are still not adjustable.

[image]

I hope there's a fix for this soon. My sidebar takes up 3/4 of my screen, and I have to keep minimizing it, which is a pain when I'm changing stuff.