bambulab/BambuStudio #7240: Avoid Crossing Walls isn't actually respected

iscofield, 2025-06-12:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4.1

### Additional system information

CPU - M4
RAM - 32GB

### Printer

Both X1C and H2D

### How to reproduce

1. Import any gridfinity box STL with a center wall
2. Check / Un-check "Avoid Crossing Walls"
3. Slice model
4. Check travel to view travel lines
5. Observe all the travel lines that cross walls, ignoring the setting

### Actual results

The sliced model is full of travels across walls, which results in absolutely horrendous print quality.

[image]

Pictures of the results of all the travels:

[image]
[image]
[image]

### Expected results

To not cross walls, which results in very poor print quality.

Using the same STL / 3MF in Orca Slicer and checking "Avoid Crossing Walls", the setting is respected:

[image]

When printing via Orcaslicer on my X1C and not having it cross walls, the print quality is flawless.

### Project file & Debug log uploads

Project File - [kitchen_gridfinity.3mf.zip](https://github.com/user-attachments/files/20700013/kitchen_gridfinity.3mf.zip)

Logs - [Archive.zip](https://github.com/user-attachments/files/20700057/Archive.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Intec-bambu, 2025-06-12:
I have started analyzing this issue. Thank you for the feedback.

iscofield, 2025-06-12:
Tried downgrading to 2.03 to see if that resolved the issue, as the newer version of BS did make changes to the "avoid crossing walls" setting, but I still see the same behavior.

iscofield, 2025-06-23:
Any updates here? The weakest part of my H2D is the fact that it has to use Bambu Slicer and I can’t use other slicers, like Orca. I’m hamstrung here as the print quality is pretty bad on this printer and I can’t do anything to fix it here.

Intec-bambu, 2025-06-23:
> Any updates here? The weakest part of my H2D is the fact that it has to use Bambu Slicer and I can’t use other slicers, like Orca. I’m hamstrung here as the print quality is pretty bad on this printer and I can’t do anything to fix it here.

Hi, apologies for the delay! This feature touches a lot of models and requires careful testing and iteration to avoid unexpected side effects. We're actively working on it and appreciate your patience in the meantime.

MikiZed, 2025-06-27:
I have the same issue, it's not just gridfinity models.
Avoid crossing walls doesn't seem to have any effect in Bambu studio 20.1.1.52

I am surprised no one faced this sooner, it's kind of a big deal... 

Orca slicer on the other hand seem to be working as expected

[image]
[image]
[image]

Intec-bambu, 2025-06-28:
> I have the same issue, it's not just gridfinity models.我也有同样的问题，不仅仅是 Gridfinity 模型。 Avoid crossing walls doesn't seem to have any effect in Bambu studio 20.1.1.52避免穿墙在 Bambu studio 20.1.1.52 中似乎没有任何效果
> 
> I am surprised no one faced this sooner, it's kind of a big deal...我很惊讶没有人更早遇到这个问题，这其实挺重要的...
> 
> Orca slicer on the other hand seem to be working as expected另一方面，Orca 切片器似乎按预期工作
> 
> [image] [image]
> 
> This is orca slicer  这是 Orca 切片器 [image]

send me the 3mf file, thanks

fhufi, 2025-07-03:
Related issue #7452 on Windows version

sakomoto987, 2025-07-21:
Hi everyone,

Just wanted to share some findings regarding the Avoid Crossing Wall feature in Bambu Studio, in case others are running into similar issues.

I’ve tested the recent 2.2.0 Public Beta, and while there are some improvements — notably, the nozzle no longer travels outside the model on curved areas — the core issue still persists: travel moves are still crossing internal walls unnecessarily, even when Avoid Crossing Wall is enabled.

You can see this clearly in the screenshots attached. These crossings occur in situations where the path planner could clearly avoid them. What makes this especially problematic is when printing flexible materials like TPU, where clean travel paths really matter for quality as mentioned.

One additional detail: I’m using a custom seam setting, but the exact same configuration works perfectly in OrcaSlicer, and was previously handled correctly in older versions of Bambu Studio.

To summarize:
	•	Bambu Studio 2.2.0 beta improves outer travel behavior on curves ✅
	•	Avoid Crossing Wall still doesn’t prevent unnecessary internal wall crossings ❌
	•	Custom seam settings may play a role, but it’s not an issue in OrcaSlicer or earlier Bambu Studio releases

Happy to share more info if needed — hoping this helps others or leads to further improvements!

[image]
[image]

Intec-bambu, 2025-08-12:
> [@Intec-bambu](https://github.com/Intec-bambu)
> 
> Update:
> 
> After several weeks of testing Bambu Studio 2.2.0 Public Beta 1, I noticed an interesting outcome. While the slicer preview still showed unexpected travel moves from the nozzle (that Avoid Crossing Wall should ideally prevent), the actual printed results were surprisingly clean. The print quality in real life was acceptable and met my expectations, despite the previewed travel paths.
> 
> However, after about a week, I moved on to testing Bambu Studio 2.2.0 Public Beta 2, and the results changed drastically — and not for the better. Using exactly the same setup (same filament, same process configuration, same model), the print quality degraded significantly. The difference is visually obvious, and I’ll attach a photo showing the issue. In Beta 2, the print output is far below the standard of Beta 1.
> 
> To confirm, I reverted back to Beta 1, and the quality immediately returned to the acceptable level I had before, again with only the known issue of unnecessary travel paths remaining.
> 
> I cannot identify what changes were introduced between Beta 1 and Beta 2 that would explain this drastic drop in print quality, but it’s clear that something in the slicing or travel path planning logic has negatively impacted the output.
> 
> Summary of findings: • Beta 1: Travel path preview still shows unnecessary moves, but real-world print quality is good/acceptable. As you can notice in attachment photo, stringing is still present even with the first beta where the nozzle travel outside the model, but acceptable at the moment.. • Beta 2: Same settings, filament, and model produce significantly worse results. • Reverting to Beta 1 restores the quality.
> 
> [@Intec-bambu](https://github.com/Intec-bambu) — would be great to know if others are seeing similar differences between Beta 1 and Beta 2, and if there’s any insight into what changed in the travel/path planning between the two builds.
> 
> Thanks, Brandon
> 
> [image]

The 'avoid crossing walls' feature has not been modified recently. However, there was a recent change related to Z-hop/retract that increases the frequency of nozzle lifts during travel to prevent collisions. I will analyze whether the TPU print quality degradation is related to this change, but my current task schedule is quite tight—please allow me some time.

By the way, reducing the Z-hop value might help minimize stringing with TPU. You can try setting Extruder->Z-hop when retract = 0 and observe if print quality improves. The underlying idea is that the nozzle attempts to wipe away oozing material across the layer surface as it moves, but this may increase the risk of collisions.

Thank you again for your valuable feedback!

Intec-bambu, 2025-08-12:
> [@Intec-bambu](https://github.com/Intec-bambu)
>
> Update:
>
> After several weeks of testing Bambu Studio 2.2.0 Public Beta 1, I noticed an interesting outcome. While the slicer preview still showed [...]

Disabling timelapse photography might also be beneficial, as the travel paths during timelapse are entirely determined by custom G-code. When timelapse is enabled, the nozzle moves to a position that maintains an unobstructed camera view after printing each layer. This may introduce unnecessary travel moves across walls

Intec-bambu, 2025-08-26:
> Have just installed version 2.2.0.85 and are experiencing the issues described above.刚刚安装了 2.2.0.85 版本，正在遇到上面描述的问题。
> 
> For me, the problem appears to occur when the infill on a layer has been completed and the nozzle returns to the start point of the next layer. During the infill process, avoid crossing walls is respected so it appears to only occur during the transition between layers.对我来说，这个问题似乎发生在某一层的填充完成后喷嘴返回到下一层起点的时候。在填充过程中，“Avoid Crossing Walls”会被遵守，所以这个问题似乎只发生在层与层之间的过渡阶段。
> 
> FYI, I have retraction set to zero for LW-PLA.供参考，我已将 LW-PLA 的回抽设为零。
> 
> Image shows avoid crossing walls being respected during infill process.图片显示“避免穿墙”在填充过程中被遵守。
> 
> <img alt="Image" width="471" height="468" src="https://private-user-images.githubusercontent.com/103159749/478820160-3a3a9edc-b0e6-4f4a-99c1-d18448cd9907.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYxNzI0MTUsIm5iZiI6MTc1NjE3MjExNSwicGF0aCI6Ii8xMDMxNTk3NDkvNDc4ODIwMTYwLTNhM2E5ZWRjLWIwZTYtNGY0YS05OWMxLWQxODQ0OGNkOTkwNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyNlQwMTM1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02M2YwZDQ4ZjYzZDZhNTY4NjQxMmRjZjQ0MjIxZTFlOWY2NmYyNzQwZTllMWZmMjQwM2U0OTU1NzA4OGJkOGM1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.t4TF6xsz9tJimN_L-CfFKXbstXMQxC_lZgiv9mfL_pA">
> Image shows the path as the nozzle returns to the layer start point图片显示了喷嘴返回到图层起点时的路径
> 
> <img alt="Image" width="1416" height="428" src="https://private-user-images.githubusercontent.com/103159749/478819699-36ae0bcd-2143-4a51-8f80-32a0bfbbecfc.PNG?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYxNzI0MTUsIm5iZiI6MTc1NjE3MjExNSwicGF0aCI6Ii8xMDMxNTk3NDkvNDc4ODE5Njk5LTM2YWUwYmNkLTIxNDMtNGE1MS04ZjgwLTMyYTBiZmJiZWNmYy5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyNlQwMTM1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kOWFhYTBkNmJjZjRjMmUwYWIyMzhjN2E2NTA1MGU3ODBiNTU4MGQ0YWM5YmYzZDYzZTYzMTUyNDhmNDNkNTg5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.8e6oJhKFUsVOG_CRdM0YxLmOgWoEZfcF2KV4llOJnHo">

This is a known issue, thank you for your feedback! Could you please package your 3MF file or other format model files into a ZIP archive and upload it as an attachment? This will help us expand our model testing dataset

Intec-bambu, 2025-08-26:
> I don't know if this information is helpful. I have the same issue when I export an object as *.amf from FreeCAD (0.21.1), but it is completely gone if I do the meshing in FreeCAD and export the object as *.stl!我不知道这些信息是否有帮助。我在从 FreeCAD (0.21.1) 将对象导出为 *.amf 时也遇到相同的问题，但如果我在 FreeCAD 中进行网格化并将对象导出为 *.stl，这个问题就完全消失了！
> 
> <img alt="Image" width="643" height="648" src="https://private-user-images.githubusercontent.com/95968772/481365616-4898d6b7-0725-43d7-a85b-7d9c68a739db.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYxNzI2MjcsIm5iZiI6MTc1NjE3MjMyNywicGF0aCI6Ii85NTk2ODc3Mi80ODEzNjU2MTYtNDg5OGQ2YjctMDcyNS00M2Q3LWE4NWItN2Q5YzY4YTczOWRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODI2VDAxMzg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAwMjIyZWMzYWQzM2RhY2VmMGI3YjE3OGVjNDQ0YTY0Y2M2ZjNmM2UxOGIzZmNmN2YwMWRiOTc4MzNkZjc1NGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2Wv3lwWHfKu8obuR-AsZa--V0RmlpjqH5bxMoDiv2uk">
> Meshed in FreeCAD  在 FreeCAD 中网格化
> 
> <img alt="Image" width="625" height="590" src="https://private-user-images.githubusercontent.com/95968772/481365621-07c60a9f-ad2b-4ffc-b3eb-681ecd05348e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYxNzI2MjcsIm5iZiI6MTc1NjE3MjMyNywicGF0aCI6Ii85NTk2ODc3Mi80ODEzNjU2MjEtMDdjNjBhOWYtYWQyYi00ZmZjLWIzZWItNjgxZWNkMDUzNDhlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODI2VDAxMzg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzZDBkOTkxZjE4NDNhN2ZkYmI1MTA1MTAzNDc0Mzg3NjU0YzQzYzljNDI2YTc2N2QxMGI1MWUxNmY3YTc1MTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2z9lVAULFyFXW8PVsJx1GEFqJiKk_EvIXAnAfR44oOM">
> Exported as *.amf  已导出为 *.amf

Could you please package your amf and stl files into a ZIP archive and upload it as an attachment?

Intec-bambu, 2025-08-26:
> Hey there, I'm new to Github and the fact that I don't know anything about coding and still talking to you shows how desperate I am.嗨，我是 GitHub 新手，而且我不会写代码，还来跟你们说话，这就说明我有多着急了。
> 
> For months now I'm trying to print a special object and I tried every possible option, settings combinations, filament drying. Nothing helps me resolve the issue.几个月来我一直在尝试打印一个特殊对象，尝试了所有可能的选项、设置组合和干燥耗材。没有任何办法能解决这个问题。
> 
> I used Orca as well to prove myself that it's not my fault and yes, same result.我也用过 Orca 来证明这不是我的错，结果也是一样。
> 
> I wanted to show you that even if the simulation shows no wall crossing what so ever, the printer DOES it. I'm hopeful now because I've seen that you talked about that bug before.我想向你展示，即使仿真显示完全没有穿墙，打印机还是会穿墙。我现在有些希望，因为我看到你之前讨论过那个漏洞。
> 
> <img alt="Image" width="1762" height="1666" src="https://private-user-images.githubusercontent.com/228377103/481766041-33ed7bee-aa7e-44a1-9bb9-9c96270dbf45.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYxNzI5MDcsIm5iZiI6MTc1NjE3MjYwNywicGF0aCI6Ii8yMjgzNzcxMDMvNDgxNzY2MDQxLTMzZWQ3YmVlLWFhN2UtNDRhMS05YmI5LTljOTYyNzBkYmY0NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgyNlQwMTQzMjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYmI4YmQxYzQxMmFhYmQ1MmNkYjZhZWNmMDk3ZDg1OTY4ZDZjNzM4NDdhZDYwMzRjYmM5ODIyOThjYmZmYjkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Bys1DtdtnOiC8mHJCGpEYGNMW75b3j7BFzzRF7jUpWk">
> The red line shows the actual path the nozzle is going.红线显示喷嘴实际移动的路径。
> 
> [image]
> 
> The print result is really poor.打印结果非常糟糕。
> 
> There is the gcode if you need it. https://we.tl/t-ufx2gNyYAU如果需要，这是 gcode。https://we.tl/t-ufx2gNyYAU
> 
> Do you need anything from me that might be helpful?你还需要我提供什么可能有帮助的信息吗？
> 
> Regards,  此致， Zirkowa

Could you please package your 3MF file or other format model files, such as stl, obj, etc., into a ZIP archive and upload it as an attachment？

nerdcraft-muc, 2025-08-26:
> Could you please package your amf and stl files into a ZIP archive and upload it as an attachment?

[DE_VOIGT_10MM-Body.zip](https://github.com/user-attachments/files/21983740/DE_VOIGT_10MM-Body.zip)

Zirkowa, 2025-08-27:
> >...
> 
> Could you please package your 3MF file or other format model files, such as stl, obj, etc., into a ZIP archive and upload it as an attachment?

Of course I can.

[stl + 3mf.zip](https://github.com/user-attachments/files/22012252/stl.%2B.3mf.zip)

Thank you very much for your effort!

Regards

dkbay, 2025-09-14:
Weirdly this doesn't seem to affect A1 printers.

Screenshot shows same avoid crossing settings and roughly same settings other than that.
But on the X1C bambu studio doesn't respect it while it looks fine on the A1.
[image]

Zirkowa, 2025-09-16:
Interesting. My case shows results from the A1. Terrible bug this is.
Luckily, I found a way to get almost perfect prints now.

Intec-bambu, 2025-09-19:
> I highly recommend to test the simulation afterwards. Look at my previous post. Bambu Studio shows perfect travelling but the actually printed result proofs that the problem is still there.
> 
> Okay, I got to check the latest version if the issue still occurs. But I doubt it.
> 
> It's by the way also existent in Orca Slicer.
> 
> Greetings!

Thanks for your feedback. I will try to review and optimize it.

RafaelEufrazio, 2025-12-06:
I'm still having this issue with the latest version. Any idea what may be happening here?

Please let me know if I'm missing something.

[image]

closed 2025-12-24 by grumpypotato-bambu