bambulab/BambuStudio #7241: Flushing volumnes for filament change not working in Windows latest version (A1 mini + AMS Lite)

thomsonOZ, 2025-06-12:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 mini

### How to reproduce

1.  Go to Prepare Tab
2.  Click on Flushing volumnes
3.  Click on Re-Calculate

### Actual results

Nothing happen in the flushing volumes

### Expected results

Flushing volumes will change based on colour

[image]

### Project file & Debug log uploads

[debug_Thu_Jun_12_12_34_36_29384.log.zip](https://github.com/user-attachments/files/20701692/debug_Thu_Jun_12_12_34_36_29384.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

thomsonOZ, 2025-06-12:
PS: Flush volumes will change when I modify the multiplier.

BoBoYu-BamBu, 2025-06-12:
@thomsonOZ  If the consumables and colors on the preparation page are the same as those for the flushing volume, once the flushing volume is opened, the flushing data will be automatically calculated. At this point, if you click "Re-calculate", the flushing volume value will not change. If you change one of the erosion values, and then click "Re-calculate", there will be a change.
[image]

BoBoYu-BamBu, 2025-06-12:
So I don't think this is a bug.

thomsonOZ, 2025-06-12:
It works again after I reboot the machine, possible my system issues