bambulab/BambuStudio #7246: Bug in Front (and maybe other) view

Peppinoc2000, 2025-06-12:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

P1S + AMS

### How to reproduce

1 open bambustudio
2 import a model 
3 click on the model
4 click on the "Move" command 
5 press ctrl+3
6 try to use arrows to actually move the object 

### Actual results

you can't move the object without changing the actual view

### Expected results

you should be able to move the model without changing view

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuDuJian-J-Zhang, 2025-06-16:
@Peppinoc2000  can not reproduce

pls take a screenshort or video

thanks

guanyun-gudujian, 2025-06-16:
@Peppinoc2000 can not reproduce

pls take a screenshort or video

thanks

Peppinoc2000, 2025-06-25:
> [@Peppinoc2000](https://github.com/Peppinoc2000) can not reproduce
> 
> pls take a screenshort or video
> 
> thanks

have you seen the video?

guanyun-gudujian, 2025-07-07:
> > [@Peppinoc2000](https://github.com/Peppinoc2000) can not reproduce
> > pls take a screenshort or video
> > thanks
> 
> have you seen the video?

yes, i have