bambulab/BambuStudio #7250: Mouse input lost

nijaba, 2025-06-12:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Additional system information

Projects for the H2D often loose mouse input after a few switch between prepare and preview.  The filament saving mode "dialog" seems to be interfering.  Only solution is to force quit Studio.

### Printer

H2D

### How to reproduce

- Create a project for the H2D
- Switch between prepare and preview adjusting some parameters
-  At some point the filament saving mode will appear and will block any mouse input into studio

### Actual results

Studio can't be used anymore, need to be force quit

### Expected results

Continuing on working

### Project file & Debug log uploads

Can't share that on a public server

### Checklist of files to include

- [ ] Log file
- [ ] Project file

nijaba, 2025-06-13:
https://github.com/user-attachments/assets/ad0dc73f-0c1b-448b-a615-1c664a10b768

Haidiye00, 2025-06-14:
Hello, thank you for recording. In the video, did you open two Bambustudio files using Bambustudio?

nijaba, 2025-06-15:
Yes

hotellonely, 2025-08-04:
Hi it's still here on newest version of BB Studio. Quite annoying...

hotellonely, 2025-09-05:
Hi, can we have any sort of update on this issue? It happens too often on macOS rendering it totally unusable.

hotellonely, 2025-09-05:
@Haidiye00 I'm counting on you QAQ

hotellonely, 2025-12-14:
请问这个问题还会有修复吗？在MacOS上用H2系列简直是个折磨，每当要切片的时候弹出这个破菜单十次有三次运气不好鼠标就再也没法正常点击了。。。能不能作为临时修复先把这个菜单改成 不要mouse over触发？？？太烦人了
[image]

logikal, 2025-12-14:
I also have this problem on MacOS! Very frustrating.

Studio 2.4.0.70
MacOS 15.6.1

XunZhangBambu, 2025-12-15:
@Von-Clausewitz @hotellonely 我们会尽快修复这个问题。目前该 hover 窗口在 Windows 平台运行正常，Mac 上出现的异常可能与平台差异有关；如果最终无法彻底修复，我们将考虑把 hover 行为调整为linux平台上点击触发的弹窗方案

hotellonely, 2025-12-15:
> [@Von-Clausewitz](https://github.com/Von-Clausewitz) [@hotellonely](https://github.com/hotellonely) 我们会尽快修复这个问题。目前该 hover 窗口在 Windows 平台运行正常，Mac 上出现的异常可能与平台差异有关；如果最终无法彻底修复，我们将考虑把 hover 行为调整为linux平台上点击触发的弹窗方案

某个网友之前说linux平台上也出现过类似问题，希望是意外。。非常感谢修复

hotellonely, 2026-02-25:
@JohnSmithCYM @lanewei120 @XunZhangBambu 请问这个问题的进展如何了。。？为啥bug修复又被revert了？bug一年了大佬们。。。

lanewei120, 2026-02-26:
That revert is on @JohnSmithCYM 's private git(https://github.com/JohnSmithCYM/BambuStudio) , not merged into the bambu's master
@eldade @hotellonely