bambulab/BambuStudio #7251: Exported models from Fusion 360 causes filament list to grow

paulelong, 2025-06-12:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

na

### Printer

Carbon X1

### How to reproduce

Use Make in fusion 360 to send a model to bambu studio

[image]

[image]

### Actual results

Every time I do this, the list of filaments grows

### Expected results

Not sure, but this isn't useful.  The list should not grow.

### Project file & Debug log uploads

easy to repro

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-13:
[image]

Hello, if you already have a lot of consumables and don't want to add them by default, you can reset them first and then click the “color match” button

paulelong, 2025-06-13:
What are "consumables"?  Is this expected behavior, or are you suggesting a temporary work around?

paulelong, 2025-06-14:
When I export the same model it adds a new filament color.  My picture shows 28, but now it's up to 32.   It just keeps on growing endlessly.   It creates a new ABS filaments based on the number of different objects.   I just want to use one of the filaments I have defined for my printer and not to have it grow each time.  

[image]

Haidiye00, 2025-06-16:
[image]

Hello, if you already have a lot of filaments and don't want to add them by default, you can reset them first and then click the “color match” button

Additionally, we will consider adding a preference setting in the future

paulelong, 2025-06-26:
I hit the RESET button, but the problem still occurs.  I don't see anyway to stop the list from growing.  I'm forced to manually delete these every once in a while, and they means clicking on each one a selecting remove.