bambulab/BambuStudio #7252: Upon slicing, Color Scheme is defaulting to the Filament option rather than Line Type

luder888, 2025-06-13:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Slice any model and in the Preview tab, the Color Scheme defaults to **Filament**, whereas in the previous version, it'd default to **Line Type**.

Please consider providing a config to customize that option.

### Actual results

Under Slicing Results, the Color Scheme defaults to **Filament**

[image]

### Expected results

In the previous version of this software, it used to default to **Line Type**

[image]

### Project file & Debug log uploads

[Bambu Cube.zip](https://github.com/user-attachments/files/20717775/Bambu.Cube.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

r4lv, 2025-06-13:
possible duplicate of https://github.com/bambulab/BambuStudio/issues/7131

JohnSmithCYM, 2025-07-31:
https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.68 We have released V2.2.0 Beta2.

closed 2025-08-01 by Haidiye00

luder888, 2025-08-01:
Please consider re-opening this issue. Last used isn't the same as defaulting it to a specific tab.

People often examine the slicing results in a sequential way. For example, someone might check the Line Type first, then check the Layer Height (if they have var layer enabled), then they might check the Speed... now it'll just default to whatever last tab was used. AND we now have to remember to change it back to Line Type if we want that to show by default next time...

luder888, 2025-08-17:
Honestly, this new "Remember last used color scheme" is just as annoying as defaulting to Filament. Now if I had used "Speed" last it always defaults to "Speed".

Please give the option to choose a SPECIFIC default tab (Filament or Line Type) so it'll behave just like how it used to be.