bambulab/BambuStudio #7254: Bambu Studio crashing on MacBook

JWeedz, 2025-06-13:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.5

### Additional system information

MacBook Pro 14-inch 2021 32GB Apple M1 Max

### Printer

Either H2D or X1C, I have both

### How to reproduce

If Bambu Studio is running (usually have a print going on) and my MacBook goes to sleep mode, the app will crash and I'll get a crash report from Apple.   

### Actual results

When restarting Bambu Studio it will ask to recover last project which has unsaved parameters.

### Expected results

It will open the last working project

### Project file & Debug log uploads

[Crash Report.3mf.zip](https://github.com/user-attachments/files/20724840/Crash.Report.3mf.zip)

[mds-2025-06-13-025646.pdf](https://github.com/user-attachments/files/20724888/mds-2025-06-13-025646.pdf)

### Checklist of files to include

- [x] Log file
- [x] Project file

XinZhangBBL, 2025-06-16:
@hansmerkl @JWeedz  Halo. We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

hansmerkl, 2025-06-16:
Here are several crashes 

[Archive.zip](https://github.com/user-attachments/files/20749135/Archive.zip)

atr2007, 2025-07-09:
Updated my original comment with the crash reports

XinZhangBBL, 2025-07-10:
Thank you for your feedback. We will resolved it at next release as soon as possible