bambulab/BambuStudio #7255: Bambu Studio crashes on Ubuntu 24.04

3D-Michael, 2025-06-13:
### Bambu Studio Version

Bambu_Studio_v02.01.00.59_24.04

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

CPU: Intel Processor N100 (Alder Lake-N, N0) 800 MHz (8.00x100.0) @ 598 MHz (6.00x99.8)
Motherboard:   WeiBu ADL-N, BIOS: 5.27, 03/27/2024, Chipset:   Intel Alder Lake-N PCH (SuperSKU ES)
Memory: 16384 MBytes @ 1330 MHz, 19-19-19-43 DDR4 SDRAM  
Graphics: Intel UHD Graphics, 7281948 KB SDRAM
Drive: CT480BX500SSD1, 468.9 GB, Serial ATA 6Gb/s @ 6Gb/s, KXG5AZNV512G TOSHIBA, 500.1 GB, NVMe
Sound: Intel Alder Lake-N PCH - cAVS (Audio, Voice, Speech)
Network: Intel Dual Band Wireless-AC 7260 AC 2x2 HMC WiFi Adapter
Network: RealTek Semiconductor RTL8168/8111 PCI-E Gigabit Ethernet NIC
OS: Ubuntu 24.04

### Printer

Bambu X1C

### How to reproduce

Error description: Bambu Studio crashes when sending data over the network to the printer's SD card. After clicking "Send," a window briefly appears with the text "Trying to connect." Bambu Studio then closes without any further error messages. This happens with the Flatpak version and also with the AppImage. This error occurs independently in all projects and is reproducible. The error did not occur with the previous version. Windows 11 is also installed on the same computer (dual boot), and this error does not occur here. I have no other problems with the network or Wi-Fi.

### Actual results

After clicking "Send," a window briefly appears with the text "Trying to connect." Bambu Studio then closes without any further error messages.

### Expected results

After clicking "Send," a window briefly appears with the text "Trying to connect." Bambu Studio then closes without any further error messages

### Project file & Debug log uploads

[Switch.zip](https://github.com/user-attachments/files/20727578/Switch.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ps68060, 2025-06-15:
I have the same.

3D-Michael, 2025-07-21:
I suspect it's related to the network plugin for Linux. The error doesn't occur on Windows, but it does occur on Linux under various (Ubuntu-based) distributions.

milk-pure, 2025-07-23:
Could you please reproduce the issue again and send us a log? The file you uploaded earlier was a 3mf file, but what we need is a log. Thank you very much for your help!

yourjelly, 2025-07-23:
here is a log where it crashes on send

[debug_Wed_Jul_23_08_28_14_341.log](https://github.com/user-attachments/files/21387948/debug_Wed_Jul_23_08_28_14_341.log)

3D-Michael, 2025-08-22:
It seems that the error isn't with BambuLab, but rather with the makers of the Flatpak version. The AppImage version and the Windows version work perfectly.

closed 2025-12-02 by BoBoYu-BamBu