bambulab/BambuStudio #7260: Bambu Studio closing repeatedly every time.

cesarmiggiolaro, 2025-06-13:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5 (24F74)

### Additional system information

Hardware Overview:

  Model Name:	MacBook Pro
  Chip:	Apple M4 Pro
  Total Number of Cores:	14 (10 performance and 4 efficiency)
  Memory:	24 GB
  System Firmware Version:	11881.121.1

### Printer

Bambu A1

### How to reproduce

Bambu Studio closing repeatedly every time.

### Actual results

I click to sent report and after that i have to reopen .

### Expected results

Restory last item opened.

### Project file & Debug log uploads

nd

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ps68060, 2025-06-14:
OS: Ubuntu 24
Bambu P1S

I upgraded on the 12th.

Every time I "send to printer" the app exits immediately.

I had to downgrade.

cesarmiggiolaro, 2025-06-14:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [41079]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               02.01.00.59 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-06-14 13:24:59.1988 -0300
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        5EDC9A64-D093-62E3-FF40-912B06904F53

Sleep/Wake UUID:       952C779A-0515-49D7-8863-3527C631FCA1

Time Awake Since Boot: 300000 seconds
Time Since Wake:       7 seconds

System Integrity Protection: enabled

Crashed Thread:        26

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Codes:       0x0000000000000001, 0x0000000000000018

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [41079]

VM Region Info: 0x18 is not in any region.  Bytes before following region: 4362649576
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10408c000-10826c000    [ 65.9M] r-x/r-x SM=COW  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x182328c34 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18233b3a0 mach_msg2_internal + 76
2   libsystem_kernel.dylib        	       0x182331764 mach_msg_overwrite + 484
3   libsystem_kernel.dylib        	       0x182328fa8 mach_msg + 24
4   CoreFoundation                	       0x182455e7c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x182454798 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x182453c58 CFRunLoopRunSpecific + 572
7   HIToolbox                     	       0x18dee827c RunCurrentEventLoopInMode + 324
8   HIToolbox                     	       0x18deeb4e8 ReceiveNextEventCommon + 676
9   HIToolbox                     	       0x18e076484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x18637bab4 _DPSNextEvent + 684
11  AppKit                        	       0x186d1a5b0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x18636ec64 -[NSApplication run] + 480
13  BambuStudio                   	       0x1074adea0 wxGUIEventLoop::OSXDoRun() + 112
14  BambuStudio                   	       0x107586b20 wxCFEventLoop::DoRun() + 40
15  BambuStudio                   	       0x107506100 wxEventLoopBase::Run() + 132
16  BambuStudio                   	       0x1074e1100 wxAppConsoleBase::MainLoop() + 120
17  BambuStudio                   	       0x107472934 wxApp::OnRun() + 36
18  BambuStudio                   	       0x107533f0c wxEntry(int&, wchar_t**) + 148
19  BambuStudio                   	       0x104f50948 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 468
20  BambuStudio                   	       0x1040a1c70 Slic3r::CLI::run(int, char**) + 33868
21  BambuStudio                   	       0x10410f32c main + 172
22  dyld                          	       0x181fcab98 start + 6076

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	       0x182328c34 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18233b3a0 mach_msg2_internal + 76
2   libsystem_kernel.dylib        	       0x182331764 mach_msg_overwrite + 484
3   libsystem_kernel.dylib        	       0x182328fa8 mach_msg + 24
4   CoreFoundation                	       0x182455e7c __CFRunLoopServiceMachPort + 160
[... 598 lines cut ...]
      },
      "deploymentId" : 240000002
    }
  ],
  "experiments" : [

  ]
}
}

Model: Mac16,8, BootROM 11881.121.1, proc 14:10:4 processors, 24 GB, SMC 
Graphics: Apple M4 Pro, Apple M4 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Mar 22 2025 02:17:00 version 23.40.30.0.41.51.180 FWID 01-989cd4a9
IO80211_driverkit-1475.39 "IO80211_driverkit-1475.39" Apr 18 2025 20:10:40
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

BoBoYu-BamBu, 2025-06-16:
Thank you for your feedback. Under what circumstances does it crash? Could you please provide the crash log? The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

ps68060, 2025-06-19:
I can confirm that the new release does not crash when sending to the printer. 

Many thanks for the quick fix.

closed 2025-07-17 by BoBoYu-BamBu