bambulab/BambuStudio #7261: Big problem with Paint feature

Christian72D, 2025-06-14:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

Core i7-1260P
32 GB RAM
Nvidia T550

### Printer

X1c

### How to reproduce

I wanted to paint a embossed text in an other color.

[https://christian72d.myds.me/downloads/download/bbs2_prepare.PNG](url)
[https://christian72d.myds.me/downloads/download/bbs2_preview.PNG](url)

### Actual results

And this is how it looks.
The text still was pure black, not like prepare or preview.
And i have white lines around the cube.

[https://christian72d.myds.me/downloads/download/IMG_20250614_054736.jpg](url)
[https://christian72d.myds.me/downloads/download/IMG_20250614_055518.jpg](url)

### Expected results

There should only be one white layer inside the word "start", not around the whole cube.

### Project file & Debug log uploads

[Box Puzzle tree support.zip](https://github.com/user-attachments/files/20735491/Box.Puzzle.tree.support.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-06-16:
Hello, I clicked on your image link but didn't see anything.
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it?  Thank you
Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif

Christian72D, 2025-06-21:
I have video of the operation.