bambulab/BambuStudio #7264: The slicer closes when sending to print.

Kastilz, 2025-06-14:
### Bambu Studio Version

Bambu_Studio_win-v02.01.00.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Printer

Neptune 4 pro

### How to reproduce

The slicer closes when sending to print.

[crash_Sat_Jun_14_20_51_59_0.log](https://github.com/user-attachments/files/20739260/crash_Sat_Jun_14_20_51_59_0.log)

### Actual results

### Expected results

[crash_Sat_Jun_14_12_02_25_0_done.log](https://github.com/user-attachments/files/20739262/crash_Sat_Jun_14_12_02_25_0_done.log)

### Project file & Debug log uploads

[boon.zip](https://github.com/user-attachments/files/20739268/boon.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

walterwongbbl, 2025-07-28:
@darkautism @Simon99de @Kastilz 
Thank you for your feedback. We will confirm this issue.

hemai249, 2025-07-28:
@Kastilz @darkautism @Simon99de Thanks for reporting this! We've passed it along to our development team, and they're working on reproducing and fixing the problem.