bambulab/BambuStudio #7266: Couldn't change the fileament sequence order

yoshio-kashiwagi, 2025-06-15:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

CPU: Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz   3.00 GHz
RAM: 32.0 GB

### Printer

Bambo lab P1S combo

### How to reproduce

Can't change the sequence order of filament with drag/drop on following dialog.
[image]

### Actual results

Couldn't change the filament order.

### Expected results

I think it could with previous version.

### Project file & Debug log uploads

No logs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-16:
@yoshio-kashiwagi Thank you for your feedback. I have tested it and found that the order of consumables can be dragged. Please refer to the attached GIF. When dragging the consumables, they need to be aligned with the replacement consumables. If that doesn't work for you, could you please try again?
[image]

yoshio-kashiwagi, 2025-06-21:
Thank you. It works fine, I had to overlap the squares.

closed 2025-06-24 by BoBoYu-BamBu