bambulab/BambuStudio #7270: Readily Crashing

Dakotawilk, 2025-06-15:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1s

### How to reproduce

Slice a model. 

Crash program

repeat a dozen times and maybe you will get through the process. 

### Actual results

Crashes, when finally slices works fine. 

### Expected results

Not Crash program.

### Project file & Debug log uploads

Couldnt find it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-16:
[image]
Hello，If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

Haidiye00, 2025-06-17:
Hello, did it not take effect after following my instructions?

JohnSmithCYM, 2025-08-14:
Can you reproduce this problem with https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85?