bambulab/BambuStudio #7273: changing nozzle size

josh211987, 2025-06-15:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 

### Additional system information

nothing relevant, this is your problem changing things that worked just fine for no reason

### Printer

all

### How to reproduce

seriously why are you making things more difficult for no reason?  Why would you take away changing the nozzle size in studio?  so now, rather than the slicer saying o you have the wrong nozzle selected and just changing it right then and there, we have to go over to the printer to do it... why?  theres no good reason for this change and it just wastes peoples time when they have printers set up in another room or even another level of the house.  thanks for continuing to be difficult

### Actual results

you wasted time programming this stupid change and now every bambu user in the world wastes more time.  thanks a lot

### Expected results

change it back, there is 0 reason i shouldnt be able to change the nozzle in the slicer

### Project file & Debug log uploads

its not a bug in the programming, its a bug in your thought process and customer service.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

slothy89, 2025-06-15:
I concur with this. I hadn't yet tested if it were firmware version related as I am still on 1.07.0.0, but having to change the Nozzle Type on a P1 with the clunky interface isn't a nice experience. Just re-enable the drop-downs on both the Handy app and Bambu Studio please.

JohnSmithCYM, 2025-06-20:
Thanks for your feedback. We have disabled nozzle setting since 2.1.0(for all printers, H2D from 2.0.3) due to the nozzle setting opeartion should be confirmed near the printer.

Asherslab, 2025-07-01:
Completely agree with @slothy89 here, doing this on the P1S is an absolute pain. 

In the very least, could you have an "advanced user" option that allows us to change the nozzle size again through studio and handy. Treating all the users as if they don't know what they're doing is a backwards way to move forward.

eried, 2025-07-17:
https://github.com/bambulab/BambuStudio/issues/7580

mytechguyri, 2025-10-16:
> Thanks for your feedback. We have disabled nozzle setting since 2.1.0(for all printers, H2D from 2.0.3) due to the nozzle setting opeartion should be confirmed near the printer.

This explains your reasoning, but it doesn't mean this was a good idea.   It is a TERRIBLE idea.  Sometimes I change my nozzle, and forget (or simply don't want to navigate through the clunky screen interface to find the setting) to change it on the printer menu itself.   There's absolutely no logical reason not to at least ALLOW the user the OPTION to update that setting within Bambu Studio.   I am a disabled individual, its very difficult for me to have to climb up 2 flights of stairs to go back to the printer just to change the nozzle type on the printer screen rather than being able to chane it right in Bambu Studio LIKE I'VE ALWAYS BEEN ABLE TO DO SINCE DAY ONE WHEN I BOUGHT MY PRINTER!  Again, you're changing core management functionality, taking away capabilities and features AFTER THE SALE.  This is unethical, predatory, and unnecessary.

eried, 2025-10-17:
They added an option to trigger nozzle detection 😁