bambulab/BambuStudio #7283: Bambu studio crashes when slicing project at 80%

D3PrintingCa, 2025-06-16:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Dell Precision 7760 Xeon W-11855m 3.2Ghz. 64GB. NVIDIA RTXA4000 / Intel UHD Graphics.

### Printer

P1S

### How to reproduce

Open step file with all default setting, slicing the plate till 80%-85% crashes every time.
Try files before which works perfectly fine but still crashing now.

[DMP log.zip](https://github.com/user-attachments/files/20748396/DMP.log.zip)

What we tried:
Set less cores, and use energy efficient way, change "cancel_glmultidraw": false to "cancel_glmultidraw" : true , 

### Actual results

All crashes as the same.

### Expected results

find the reason and fix the issue for work.

### Project file & Debug log uploads

[log_iotc.txt](https://github.com/user-attachments/files/20748430/log_iotc.txt)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-06-16:
[image]
Hello，If you still encounter slicing crashes, please close the software first, then find this file through Everything software, and change "cancel_glmultidraw": false to "cancel_glmultidraw" : true   , and finally start the software

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, please update the software version.
After installation, if the slice crashes, please find "Open Configuration Folder" in the Help menu, then close the software and modify the configuration file's "false" to "true" as shown in the figure below. Reopen the software and help check if the slice still crashes. Thank you~

[image]

[image]