bambulab/BambuStudio #7286: Bambu Lab crashes after slicing before preview mode

D3PrintingCa, 2025-06-16:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

W11

### Additional system information

Dell precision 7760 Xeon W11855M3.2HA 64GB NVDIA4000

### Printer

P1S

### How to reproduce

Import the stp file with default setting
Slicing the project and crashed around 80%.
Every time.

### Actual results

Bambu Studio crashes every time.

### Expected results

Find out the issue and fix the bug.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20748509/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

D3PrintingCa, 2025-06-16:
So far, try less cores (down from 12 to11, 6, 4, 3, 2 and even 1) / change "cancel_glmultidraw": false to "cancel_glmultidraw" : true
None of them works.

Haidiye00, 2025-06-16:
hello, can you provide your 3mf

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, please update the software version.
After installation, if the slice crashes, please find "Open Configuration Folder" in the Help menu, then close the software and modify the configuration file's "false" to "true" as shown in the figure below. Reopen the software and help check if the slice still crashes. Thank you~

[image]

[image]