bambulab/BambuStudio #7291: Bambu studio for Mac crashes if I leave it running and put my machine to sleep

neilcsinger, 2025-06-16:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.5

### Additional system information

MacBook Pro, Apple M4 Pro, 48 GB

### Printer

Bambu Labs P1P

### How to reproduce

Bug started with latest update.  It never happened before.
1. Print with bambu studio
2. wait for download and printer to start
3. put Mac to sleep by closing its cover
4. open Mac and Bambu Studio is crashed

### Actual results

Bambu Studio crashes

### Expected results

It should still be running (as in the previous versions)

### Project file & Debug log uploads

[Outer Wheel 2.3mf.zip](https://github.com/user-attachments/files/20749984/Outer.Wheel.2.3mf.zip)
[log files.zip](https://github.com/user-attachments/files/20749983/log.files.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

AndriyDmytrenko, 2025-06-20:
Still reproduces for me.
Bambu Studio 2.1.1.52

perlbo, 2025-06-23:
Same here. New plugin does not solve the problem

neilcsinger, 2025-06-23:
Yes. It still crashes for me as well. V2.1.1.52

TinyMito, 2025-06-24:
> [@neilcsinger](https://github.com/neilcsinger) Thank you for your feedback. We have already fixed this issue and released a new plugin: V2.1.0.56. Please update the plugin and try again.

How do I verify if I have the latest plugin? I have ran check for updates, nothing shows up and I assume I am already on the latest version. It is still crashing.

perlbo, 2025-06-25:
> [@TinyMito](https://github.com/TinyMito) [@neilcsinger](https://github.com/neilcsinger) [@perlbo](https://github.com/perlbo) [@AndriyDmytrenko](https://github.com/AndriyDmytrenko), Thank you for your feedback. Could you please copy and paste the crash report of MAC for us.

[crashj report.pdf](https://github.com/user-attachments/files/20894216/crashj.report.pdf)

XinZhangBBL, 2025-06-26:
@perlbo Thank you.  From the report, i recognize that the process freezed not crashed. Is that true?

[image]

FyrbyAdditive, 2025-06-26:
I also have this problem, same spec/OS version as the OP. It crashes reliably every night without fail.

XinZhangBBL, 2025-06-26:
@perlbo Thank you for your feedback. Your issue will be fixed on next plugin version.
@FyrbyAdditive @neilcsinger @AndriyDmytrenko @TinyMito  Could you please copy and paste the crash or freeze report of MAC for us.

FyrbyAdditive, 2025-07-02:
Sorry for the delay. Here it is:

[bbs crash.txt](https://github.com/user-attachments/files/21015525/bbs.crash.txt)

Plastdroid, 2025-09-18:
Issue persists as of v.02.2.2.56 , intel mac. Sequoia 15.6.1

closed 2025-12-02 by BoBoYu-BamBu

timmead, 2026-01-05:
Yeah, it still happens for me too.