bambulab/BambuStudio #7303: H2D not showing in Multi-Device View

K2Kevin, 2025-06-16:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.5

### Additional system information

H2D not showing in Multi-Device View

### Printer

H2D

### How to reproduce

Open Multi-Device View

### Actual results

Missing from Edit Printers and list.

### Expected results

Expect to be able to select the H2D

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-17:
@K2Kevin Thank you for your feedback. Currently, the H2D model does not support multi-device functionality.

fubbi, 2026-02-04:
crickets...

gshlomi, 2026-02-04:
@BoBoYu-BamBu why is this issue closed? Still not fixed...
Can we get an update on the request?