bambulab/BambuStudio #7308: 局域网模式下整个软件非常卡顿及局域网模式连接机器不成功

afreeYF, 2025-06-17:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 家庭中文版 22H2

### Additional system information

cpu:    AMD Ryzen 7 5800U with Radeon Graphics            1.90 GHz
16.0 GB (13.9 GB 可用)
64 位操作系统, 基于 x64 的处理器

### Printer

P1S

### How to reproduce

1，工作室连接的路由器网络经常不好，通常使用局域网模式发送和打印，机器和电脑在同一个局域网中。
2，连接的是同一个的网络，但是wifi本身是不可访问internet的，软件好像会一直请求判断，任何切片操作都会一卡一卡的
3，机器开启局域网模式，输入访问码，总是连接不成功，
4，如果一开始路由器突然没有网络的情况下，软件的设备列表也能看到所有机器，但是点击之后让输入访问码，
5，输入后说连接中，也不会提示访问码错误，然后过一会连接的那个弹窗消失，也没有任何提示成功和失败，
6，再次点击列表中的机器还让输入访问码
7，整个软件一会卡死，一会卡死，直接软件崩掉，windows提示让关闭

### Actual results

1，工作室连接的路由器网络经常不好，通常使用局域网模式发送和打印，机器和电脑在同一个局域网中。
2，连接的是同一个的网络，但是wifi本身是不可访问internet的，软件好像会一直请求判断，任何切片操作都会一卡一卡的
3，机器开启局域网模式，输入访问码，总是连接不成功，
4，如果一开始路由器突然没有网络的情况下，软件的设备列表也能看到所有机器，但是点击之后让输入访问码，
5，输入后说连接中，也不会提示访问码错误，然后过一会连接的那个弹窗消失，也没有任何提示成功和失败，
6，再次点击列表中的机器还让输入访问码
7，整个软件一会卡死，一会卡死，直接软件崩掉，windows提示让关闭

### Expected results

期待局域网下能正常连接和发送文件，
说句不该说的，我的其他创想三维的机器，wifi无internet访问时，局域网也是直接连接，发送文件也很快，有网的时候也是无感直接发送和使用。
就拓竹切片软件一没网整个要卡死啦，连切片操作界面中都一卡一卡的，机器开了局域网模式，输入访问码也是连接不成功，工作室一断网这个机器和 软件跟废了一样。

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20769259/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

afreeYF, 2025-06-17:
哦哦，谢谢回复，哪里可以下到lan  only版本呢，求链接，非常感谢

---- 回复的原邮件 ----
| 发件人 | Freeman ***@***.***> |
| 日期 | 2025年06月17日 17:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [bambulab/BambuStudio] 局域网模式下整个软件非常卡顿及局域网模式连接机器不成功 (Issue #7308) |
hotellonely left a comment (bambulab/BambuStudio#7308)

拓竹云设计就是这样的。如果你想要比较稳定的本地连接只能使用Lan only模式

hotellonely, 2025-06-18:
> 哦哦，谢谢回复，哪里可以下到lan  only版本呢，求链接，非常感谢
> […](#)

退出bambu studio里的拓竹账号，然后本地连接打印机，打印机打开局域网模式

JohnSmithCYM, 2025-07-09:
感谢您的反馈，我们已经将该需求放入了需求池中，后续考虑在云端弱网的情况下优化相关体验。

Mujin-code, 2026-08-18:
@Cryghast 您好，感谢你的反馈，您是否尝试过最新的版本；是否还会出现一样的场景下连接失败的情况，如果还会出现，是否可以给我们提供一下您的日志信息呢。