bambulab/BambuStudio #7313: Display settings

KurtaliasdiabloCH, 2025-06-17:
### Bambu Studio Version

2.1.0.59

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

HP ZBook Studio 16 inch G10 Mobile workstation PC
13th Gen Intel(R) Core(TM) i9-13900HK
32.0 Go (31.6 Go utilisable)

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Set resolution screen to 3840 x 2400 (native resolution)
Set screen scaling to 250% (as screen size is 16 inch, 100% scaling makes UI too small)

### Actual results

In Bambu Studio, the left part of the UI is not adjusted, because too thin, so UI is not usable and parameters part of UI is not adjustable in the width.

### Expected results

I want the UI to be the best and it do not depend on resolution and scaling settings.

### Project file & Debug log uploads

It doesn't need any files for this issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-18:
https://github.com/Haidiye00/BambuStudio/actions/runs/15534911769
[image]
This is a temporary version, please try it out~thanks

1.open this zip software,come to preivew tab, then close software. find this BambuStudio.conf,and modify enable_sidebar_resizable 's value
[image]
2.reopen this  zip software, come to preivew tab,then press "ctrl+w"
[image]

then you can adjust sidebar's width

Soygen, 2025-06-18:
That works for me.

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

Hello, please note that after opening the software, it is recommended that you press Ctrl+W before adjusting the width