bambulab/BambuStudio #7319: Wiki update for Ubuntu build / install guide

mxmlnkn, 2025-06-17:
As I cannot download the releases because of a 503 service unavailable error, I tried to build from source. The linked wiki [here](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide) seems to be outdated.

The install command fails to find `libwebkit2gtk-4.0-dev` because Ubuntu 24.04 only has `libwebkit2gtk-4.1-dev`. I'd recommend using a glob to make this future-safe `libwebkit2gtk-4.*-dev`.

The very next step, calling cmake to build the deps, fails because it cannot find boost, which seems to have been forgotten in the install instructions.

```
CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.28/Modules/FindBoost.cmake:2393 (find_package_handle_standard_args)
  CMakeLists.txt:361 (find_package)

-- Configuring incomplete, errors occurred!
```

After installing boost system filesystem thread, there are lots of more missing dependencies... Here is the full list of missing dependencies:

Therefore, add `libboost-filesystem-dev libboost-system-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libboost-iostreams-dev libcurlpp-dev libglew-dev libglfw3-dev libcereal-dev libnlopt-dev libnlopt-cxx-dev` to the apt-get command at the top.

I am stuck at CMake looking for OpenVDB:

```
-- OpenVDB ABI Version: 10
CMake Warning at cmake/modules/FindOpenVDB.cmake:346 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:368 (just_fail)
  CMakeLists.txt:570 (find_package)

CMake Error at CMakeLists.txt:575 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
```

A simple `sudo apt-get install libopenvdb-dev libopenvdb-ax-dev` did not work. Even `sudo apt install *openvdb*` does not help.

Please update the build instructions.

jarczakpawel, 2025-06-19:
I successfully built and ran BambuStudio on Debian 13 (trixie).

System details:

Debian GNU/Linux 13 (trixie)

Kernel: 6.12.32-amd64 (PREEMPT_DYNAMIC, June 2025)

I installed all dependencies using apt, including required development packages like libwebkit2gtk-4.1-dev, libwxgtk3.2-dev, Boost, OpenVDB, CGAL, and others.

sudo apt install \
  git cmake clang g++ build-essential m4 pkgconf \
  libgl1-mesa-dev libglu1-mesa-dev libosmesa6-dev \
  libwayland-dev libxkbcommon-dev wayland-protocols \
  extra-cmake-modules \
  libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.1-dev \
  libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
  gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad \
  nasm yasm libx264-dev \
  libboost-filesystem-dev libboost-system-dev libboost-thread-dev \
  libboost-log-dev libboost-locale-dev libboost-iostreams-dev \
  libcurlpp-dev libglew-dev libglfw3-dev \
  libcereal-dev libnlopt-dev libnlopt-cxx-dev \
  libopenvdb-dev libopenvdb-tools \
  libwxgtk3.2-dev \
  libcgal-dev libopencv-dev \
  libocct-foundation-dev libocct-data-exchange-dev libocct-ocaf-dev \
  libocct-modeling-data-dev libocct-modeling-algorithms-dev \
  libocct-visualization-dev
  
  
  
cd ~/projects/bambustudio/deps
mkdir -p build && cd build

cmake .. \
  -DDESTDIR="$HOME/projects/bambustudio_dep" \
  -DCMAKE_BUILD_TYPE=Release \
  -DDEP_WX_GTK3=1 \
  -DOPENVDB_FIND_MODULE_PATH=/usr/lib/x86_64-linux-gnu/cmake/OpenVDB \
  -DJPEG_VERSION=8

make -j"$(nproc)"

mkdir -p ~/projects/bambustudio_build
mkdir -p ~/projects/bambustudio_install
cd       ~/projects/bambustudio_build

cmake ../bambustudio \
  -DSLIC3R_STATIC=ON \
  -DSLIC3R_GTK=3 \
  -DBBL_RELEASE_TO_PUBLIC=1 \
  -DCMAKE_PREFIX_PATH="$HOME/projects/bambustudio_dep/usr/local" \
  -DCMAKE_INSTALL_PREFIX="$HOME/projects/bambustudio_install" \
  -DCMAKE_BUILD_TYPE=Release

cmake --build . --target install --config Release -j"$(nproc)"

I run it like this:

env QT_QPA_PLATFORM=xcb /home/berlin/projects/bambustudio_install/bin/bambu-studio

mxmlnkn, 2025-06-19:
Thank you for the help! It worked. Embarrassingly, I probably was in the wrong folder and tried to run CMake in the top-level instead of inside `deps` :(.

closed 2025-06-19 by mxmlnkn