bambulab/BambuStudio #7322: 日本語の文字入力が実行できない

Jun-ji-Jun-ji, 2025-06-18:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

mac os 15.5

### Additional system information

mac os sequoia 15.5

### Printer

A1

### How to reproduce

1,bambustudio新作成ープリミティブを追加する
2、テキスト入力コマンドを選択、テキストを入力できるようにする
3、日本語フォントを選択しても、「テキスト」の入力欄が日本語にならない

[image]

### Actual results

言語入力切替操作を行っても切り替わらず、英語入力のままになってしまう

### Expected results

言語入力切り替え操作を行ったら、日本語入力に切り替わること

### Project file & Debug log uploads

[ディスク.3mf.zip](https://github.com/user-attachments/files/20788964/3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Jun-ji-Jun-ji, 2025-06-19:
お返事ありがとうございます。感謝します。

日本語の文字をペーストしても、良い結果にはなりませんでした。

公式回答を頂いたので共有致します。

------
日本語入力の件についてですが、大変恐れ入りますが、現在のBambu Studioバージョンでは、日本語テキスト入力に正式対応しておりません。そのため、テキスト入力時に日本語が正しく反映されない、あるいは入力できない現象が発生いたします。

本件につきましては、既に開発元にも要望として報告済みであり、今後のアップデートにて対応が検討される予定です。ただし、現時点では対応時期は未定でございます。
------

Haidiye00, 2025-07-02:
[image]
こんにちは、最新の内部バージョンは解決しました。次のバージョンがリリースされます~お楽しみに
Hello, our latest internal version has been resolved and the next version will be released~Stay tuned

Jun-ji-Jun-ji, 2025-07-02:
thank you so much.

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

closed 2025-07-17 by Haidiye00

Haidiye00, 2025-07-18:
Hello, can you download QuickRecorder software and record a GIF of the operation video and upload it? Thank you                         [lihaoyun6/QuickRecorder: A lightweight screen recorder based on ScreenCapture Kit for macOS / 基于 ScreenCapture Kit 的轻量化多功能 macOS 录屏工具](https://github.com/lihaoyun6/QuickRecorder/releases/tag/1.6.5)

For Japanese on Mac, due to some input issues with imgui,
We suggest that you enter it in your diary and then copy and paste it into our software. Thank you

Haidiye00, 2025-07-18:
For Japanese on Mac, due to some input issues with imgui,
We suggest that you enter it in your diary and then copy and paste it into our software. Thank you

Jun-ji-Jun-ji, 2025-07-19:
> Mac の日本語の場合、imgui の入力に問題があるため、日記や他のソフトウェアに入力し、コピーして当社のソフトウェアに貼り付けることをお勧めします。ありがとう

日本語をペーストして入力する事が出来ました。ありがとうございます。