bambulab/BambuStudio #7327: Can't connect to printer with Bambu Studio while using Bambu Farm Manager

JonasKuschel, 2025-06-18:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS/Also tested on Windows 11

### Additional system information

M1 Pro
MacBook 2021

### Printer

Bambu Lab P1S

### How to reproduce

I have setup Bambu Print Farm Manager on my windows server. I connected all of my printers to the farm manager. Afterwards I have tried to connect bambu studio to the printers as well. But I get the Error: "Already bound to another device" or similar. Actually I had this issue in the beginning where I saw the printer in Bambu Studio under "other devices" and when I clicked it I got the error. Now when I try to connect a new printer by Access code I click on OK and nothing happens.

### Actual results

it is not possible to bin it.

### Expected results

I expect bambu studio to be bound to my printers even though they are connected to bambu farm manager.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JonasKuschel, 2025-06-18:
[image]

JonasKuschel, 2025-06-19:
No it is not bound to handy it is bound to the farm. However, if it is bound to the farm it is not possible to bind it to Bambu Studio. Binding to the farm has never been an issue. 

In my recent connections I tried using my Mac to connect the printer (which is connected to the farm) to the mac version of Bambu Studio. Now I have switched to my Windows PC and tried again still no result. What has changed however is visible in this picture

[image]

It shows this message in the bottom and closes afterwards. I never get to the point to insert the serial number.

Still Bambu Studio is not connected to the Printer.

JohnSmithCYM, 2025-06-20:
Thank you very much for your feedback! Could you provide a log file of Studio for us to analyze and help you solve the problem?

[image]

[image]

JonasKuschel, 2025-08-03:
I still have the issue. I would really like to switch my farm to Bambu Farm Manager. But without the Bambu Slicer connection I can't troubleshoot nor develop new designs.

closed 2025-10-19 by JonasKuschel