bambulab/BambuStudio #7332: windows sever 2016无法打开最新版本安装包

Finehappily, 2025-06-19:
### Bambu Studio Version

02.01.00.59

### Where is the application from?

Bambu Lab Official website

### OS version

windows sever 2016 standard 1607 14393.8148

### Additional system information

intel 12300T
8G DDR4

### Printer

PIS

### How to reproduce

从02.01.00.59更新到02.01.01.52，无法打开02.01.01.52的安装包，改成兼容win7启动也不行

### Actual results

打不开02.01.01.52的安装包

### Expected results

安装02.01.01.52版本

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-06-24:
@Finehappily 

您好，打不开的现象是啥？能录一个简短的视频么

另外，确认是只有2.1版本才出现吗

Finehappily, 2025-06-24:
https://github.com/user-attachments/assets/c6c0c908-741e-4790-b524-1482f63074a0
再长一点的视频发不出来，总之就是老版本能打开，新版本的打不开

Finehappily, 2025-06-24:
我还是装免安装版的吧
1defender我一直是关的
2安装包下过很多次了
3重启以后安装也一样