bambulab/BambuStudio #7338: [Wayland][NVIDIA] Fix black/empty preview by enabling Zink-based hardware rendering

jarczakpawel, 2025-06-19:
**Is your feature request related to a problem? Please describe.**
Running Bambu Studio / Orca Slicer on Wayland with an NVIDIA GPU (driver ≥ 555, Debian Testing/Sid) shows only an empty grey/black viewport – the Home tab loads but the model preview is missing.
The same happens with the upstream AppImage and Flatpak builds, so it is not a packaging bug.

**Describe the solution you'd like**
Leverage Mesa Zink to bypass the flaky NVIDIA OpenGL stack while still keeping full GPU acceleration:

`apt install vulkan-tools mesa-vulkan-drivers`

My bin for the compiled program is in the path: ~/projects/bambustudio_install/bin/bambu-studio
`#!/usr/bin/env bash

export GDK_BACKEND=wayland
export QT_QPA_PLATFORM=wayland

# ←––– KLUCZOWE –––→
export __GLX_VENDOR_LIBRARY_NAME=mesa
export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
export MESA_LOADER_DRIVER_OVERRIDE=zink
export GALLIUM_DRIVER=zink
export WEBKIT_DISABLE_DMABUF_RENDERER=1

exec "$HOME/projects/bambustudio_install/bin/bambu-studio" "$@"`

`chmod +x ./bambu-zink.sh`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
System where it works 100 %
• Debian GNU/Linux 13 (trixie/testing)
• Kernel 6.12.32-amd64
• NVIDIA driver 575.57 (beta .run installer)
• mesa-vulkan-drivers + vulkan-tools from Debian Testing

**How I compiled it:**

sudo apt install
git cmake clang g++ build-essential m4 pkgconf
libgl1-mesa-dev libglu1-mesa-dev libosmesa6-dev
libwayland-dev libxkbcommon-dev wayland-protocols
extra-cmake-modules
libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.1-dev
libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad
nasm yasm libx264-dev
libboost-filesystem-dev libboost-system-dev libboost-thread-dev
libboost-log-dev libboost-locale-dev libboost-iostreams-dev
libcurlpp-dev libglew-dev libglfw3-dev
libcereal-dev libnlopt-dev libnlopt-cxx-dev
libopenvdb-dev libopenvdb-tools
libwxgtk3.2-dev
libcgal-dev libopencv-dev
libocct-foundation-dev libocct-data-exchange-dev libocct-ocaf-dev
libocct-modeling-data-dev libocct-modeling-algorithms-dev
libocct-visualization-dev

cd ~/projects/bambustudio/deps
mkdir -p build && cd build

cmake ..
-DDESTDIR="$HOME/projects/bambustudio_dep"
-DCMAKE_BUILD_TYPE=Release
-DDEP_WX_GTK3=1
-DOPENVDB_FIND_MODULE_PATH=/usr/lib/x86_64-linux-gnu/cmake/OpenVDB
-DJPEG_VERSION=8

make -j"$(nproc)"

mkdir -p ~/projects/bambustudio_build
mkdir -p ~/projects/bambustudio_install
cd ~/projects/bambustudio_build

cmake ../bambustudio
-DSLIC3R_STATIC=ON
-DSLIC3R_GTK=3
-DBBL_RELEASE_TO_PUBLIC=1
-DCMAKE_PREFIX_PATH="$HOME/projects/bambustudio_dep/usr/local"
-DCMAKE_INSTALL_PREFIX="$HOME/projects/bambustudio_install"
-DCMAKE_BUILD_TYPE=Release

cmake --build . --target install --config Release -j"$(nproc)"

marcszy91, 2025-06-19:
I have the same Problem on my Macbook M1 Pro with MacOs 15.5 can you help me?

oe3gwu, 2025-06-19:
> **Is your feature request related to a problem? Please describe.** Running Bambu Studio / Orca Slicer on Wayland with an NVIDIA GPU (driver ≥ 555, Debian Testing/Sid) shows only an empty grey/black viewport – the Home tab loads but the model preview is missing. The same happens with the upstream AppImage and Flatpak builds, so it is not a packaging bug.
> 
> **Describe the solution you'd like** Leverage Mesa Zink to bypass the flaky NVIDIA OpenGL stack while still keeping full GPU acceleration:
> 
> `apt install vulkan-tools mesa-vulkan-drivers`
> 
> My bin for the compiled program is in the path: ~/projects/bambustudio_install/bin/bambu-studio
> 
> > #!/usr/bin/env bash
> > export GDK_BACKEND=wayland
> > export QT_QPA_PLATFORM=wayland
> > export __GLX_VENDOR_LIBRARY_NAME=mesa
> > export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json
> > export MESA_LOADER_DRIVER_OVERRIDE=zink
> > export GALLIUM_DRIVER=zink
> > export WEBKIT_DISABLE_DMABUF_RENDERER=1
> > exec "$HOME/projects/bambustudio_install/bin/bambu-studio" "$@"
> 
> `chmod +x ./bambu-zink.sh`
> 
> **Describe alternatives you've considered** I tried multiple workarounds before finding a proper solution. I tested both the Flatpak and AppImage versions of BambuStudio/OrcaSlicer, but the model preview pane remained black or empty. I also experimented with downgrading libwebkit2gtk and other related GTK/WebKit libraries, but that did not help either. None of these alternatives solved the rendering issue on Wayland with NVIDIA.
> 
> **Additional context** System where it works 100 % • Debian GNU/Linux 13 (trixie/testing) • Kernel 6.12.32-amd64 • NVIDIA driver 575.57 (beta .run installer) • mesa-vulkan-drivers + vulkan-tools from Debian Testing
> 
> **How I compiled it:**
> 
> sudo apt install git cmake clang g++ build-essential m4 pkgconf libgl1-mesa-dev libglu1-mesa-dev libosmesa6-dev libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.1-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad nasm yasm libx264-dev libboost-filesystem-dev libboost-system-dev libboost-thread-dev libboost-log-dev libboost-locale-dev libboost-iostreams-dev libcurlpp-dev libglew-dev libglfw3-dev libcereal-dev libnlopt-dev libnlopt-cxx-dev libopenvdb-dev libopenvdb-tools libwxgtk3.2-dev libcgal-dev libopencv-dev libocct-foundation-dev libocct-data-exchange-dev libocct-ocaf-dev libocct-modeling-data-dev libocct-modeling-algorithms-dev libocct-visualization-dev
> 
> cd ~/projects/bambustudio/deps mkdir -p build && cd build
> 
> cmake .. -DDESTDIR="$HOME/projects/bambustudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1 -DOPENVDB_FIND_MODULE_PATH=/usr/lib/x86_64-linux-gnu/cmake/OpenVDB -DJPEG_VERSION=8
> 
> make -j"$(nproc)"
> 
> mkdir -p ~/projects/bambustudio_build mkdir -p ~/projects/bambustudio_install cd ~/projects/bambustudio_build
> 
> cmake ../bambustudio -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="$HOME/projects/bambustudio_dep/usr/local" -DCMAKE_INSTALL_PREFIX="$HOME/projects/bambustudio_install" -DCMAKE_BUILD_TYPE=Release
> 
> cmake --build . --target install --config Release -j"$(nproc)"

Your solution works! Thanks! Non the less, this bug should be solved.!

oe3gwu, 2025-06-19:
> I have the same Problem on my Macbook M1 Pro with MacOs 15.5 can you help me?

My Macbook M1 with latest MacOS works fine. This was my fallback solution, because I refuse to use Windows.

gavonmothma, 2025-07-05:
I finally managed to find a solution to this for the flatpak

```
flatpak override --user \
  --env=__EGL_VENDOR_LIBRARY_FILENAMES=/usr/lib/x86_64-linux-gnu/GL/glvnd/egl_vendor.d/50_mesa.json \
  --env=GALLIUM_DRIVER=zink \
  --env=MESA_LOADER_DRIVER_OVERRIDE=zink \
  --env=WEBKIT_DISABLE_DMABUF_RENDERER=1 \
  com.bambulab.BambuStudio
```

Then you can launch Bambu Studio as you normally would. If you want to remove this later, you can use
`flatpak override --reset com.bambulab.BambuStudio`

To verify
`flatpak override --show com.bambulab.BambuStudio`

asheiduk, 2025-08-13:
@gavonmothma The `--show` and `--reset` commands are missing the `--user` option used by the first command.