bambulab/BambuStudio #7343: No 3D Visualization

oe3gwu, 2025-06-19:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab github releases

### OS version

OS: Kubuntu 25.04

### Additional system information

```
           `.:/ossyyyysso/:.                rainer@hyperion
        .:oyyyyyyyyyyyyyyyyyyo:`            ---------------
      -oyyyyyyyodMMyyyyyyyysyyyyo-          OS: Kubuntu x86_64
    -syyyyyyyyyydMMyoyyyydmMMyyyyys-        Kernel: Linux 6.14.0-15-generic
   oyyysdMysyyyydMMMMMMMMMMMMMyyyyyyyo      Uptime: 42 mins
 `oyyyydMMMMysyysoooooodMMMMyyyyyyyyyo`     Packages: 3535 (dpkg), 30 (snap)
 oyyyyyydMMMMyyyyyyyyyyyysdMMysssssyyyo     Shell: bash 5.2.37
-yyyyyyyydMysyyyyyyyyyyyyyysdMMMMMysyyy-    Display (DELL U2419H): 1920x1080 @ 60 Hz in 24" [External] *
oyyyysoodMyyyyyyyyyyyyyyyyyyydMMMMysyyyo    Display (DELL U2419H): 1920x1080 @ 60 Hz in 24" [External]
yyysdMMMMMyyyyyyyyyyyyyyyyyyysosyyyyyyyy    DE: KDE Plasma 6.3.4
yyysdMMMMMyyyyyyyyyyyyyyyyyyyyyyyyyyyyyy    WM: KWin (Wayland)
oyyyyysosdyyyyyyyyyyyyyyyyyyydMMMMysyyyo    WM Theme: Breeze
-yyyyyyyydMysyyyyyyyyyyyyyysdMMMMMysyyy-    Theme: Breeze (Dark) [Qt], Breeze-Dark [GTK2], Breeze [GTK3/4]
 oyyyyyydMMMysyyyyyyyyyyysdMMyoyyyoyyyo     Icons: breeze-dark [Qt], breeze-dark [GTK2/3/4]
 `oyyyydMMMysyyyoooooodMMMMyoyyyyyyyyo      Font: Noto Sans (10pt) [Qt], Noto Sans (10pt) [GTK2/3/4]
   oyyysyyoyyyysdMMMMMMMMMMMyyyyyyyyo       Cursor: breeze (24px)
    -syyyyyyyyydMMMysyyydMMMysyyyys-        Terminal: konsole 24.12.3
      -oyyyyyyydMMyyyyyyysosyyyyo-          CPU: AMD Ryzen 5 5600X (12) @ 4.65 GHz
        ./oyyyyyyyyyyyyyyyyyyo/.            GPU: NVIDIA GeForce RTX 3060 Lite Hash Rate [Discrete]
           `.:/oosyyyysso/:.`               Memory: 5.95 GiB / 31.24 GiB (19%)
                                            Swap: Disabled
                                            Disk (/): 253.93 GiB / 456.89 GiB (56%) - ext4
                                            Disk (/mnt/daten): 1.61 TiB / 5.82 TiB (28%) - xfs
                                            Disk (/mnt/spiele): 115.48 GiB / 915.82 GiB (13%) - ext4
                                            Disk (/mnt/temp): 111.38 GiB / 1.79 TiB (6%) - ext4
                                            Local IP (enp8s0): 192.168.1.131/24
                                            Locale: de_AT.UTF-8

```

### Printer

Bambu P1S

### How to reproduce

1. Start the AppImage provided for Ubuntu 24.04
2. Go to preview and see the empty screen

### Actual results

U see that the 3D Visualization is missing

[image]

### Expected results

3D View should be present

### Project file & Debug log uploads

[log.tar.gz](https://github.com/user-attachments/files/20823880/log.tar.gz)

### Checklist of files to include

- [x] Log file
- [ ] Project file

jarczakpawel, 2025-06-19:
https://github.com/bambulab/BambuStudio/issues/7338

oe3gwu, 2025-06-29:
> This is a duplicate of several bugs: [#6632](https://github.com/bambulab/BambuStudio/issues/6632), [#6985](https://github.com/bambulab/BambuStudio/issues/6985), [#6066](https://github.com/bambulab/BambuStudio/issues/6066) and there are several work-arounds in those threads and in [#7338](https://github.com/bambulab/BambuStudio/issues/7338), depending on if you use the AppImage or Flatpak. It comes down to NVIDIA + Wayland

Honestly I havent read through all the bugs, but this software is the only one I use that has issues with Wayland + NVIDIA.

guanyun-gudujian, 2025-07-03:
> Honestly I havent read through all the bugs, but this software is the only one I use that has issues with Wayland + NVIDIA.

yes, as BS do not support wayland natively
for now we fallback to x11 mode on wayland platform as a workround . will be released soon
for more info, pls see https://github.com/bambulab/BambuStudio/issues/6985#issuecomment-2918076910

guanyun-gudujian, 2025-07-18:
@oe3gwu 
Hi, the offical package with the x11 mode workaround has been released

pls have a try

thanks

https://github.com/bambulab/BambuStudio/releases

guanyun-gudujian, 2025-07-31:
> It's still empty for me in Fedora 42 with 2.2.0.60 :(
> 
> <img alt="Image" width="2000" height="1682" src="https://private-user-images.githubusercontent.com/57287657/471069344-59901ecd-e39f-4a47-b13e-a0d529f7870e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM5MjUxNzYsIm5iZiI6MTc1MzkyNDg3NiwicGF0aCI6Ii81NzI4NzY1Ny80NzEwNjkzNDQtNTk5MDFlY2QtZTM5Zi00YTQ3LWIxM2UtYTBkNTI5Zjc4NzBlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzMxVDAxMjExNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNkN2JjZjJlYjBmMzAwYzEwYWIyNGE5MmJiMzY3Y2YyZmY2YjcyZDAwNmNiMjcwNjIyNDc5MmYwYzJiYTdiZTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cvu1SKS266VLME6XB5GNat2y6lZu_0ERvjRTsJOstUk">
> However it works with the workaround described in [#7338](https://github.com/bambulab/BambuStudio/issues/7338)

@Mr-MooMoo 
pls help to check which one work, thanks

version1 : (without platform checking)
for ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562148808
for ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366488248/artifacts/3562130141

version2: (with platform checking)
for ubuntu 24.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562161529
for ubuntu 22.04: https://github.com/guanyun-gudujian/BambuStudio/actions/runs/16366503233/artifacts/3562154867

guanyun-gudujian, 2025-07-31:
> I am on Fedora, sry :[

just a linux version

can run on Fedora, pls have a try, thanks

GuDuJian-J-Zhang, 2025-08-01:
> Sorry for me being a total noob here, but I have absolutely no experience with c/c++ 🙁 ;
> 
> * How can I compile the links you gave me?

@Mr-MooMoo 
no need to compile

just click the link to download the package

Mr-MooMoo, 2025-08-07:
I am on vacation at the moment, I ll be back at home next week and will test it immediately!

guanyun-gudujian, 2025-08-08:
> Hello! The newest release is now working for me on Fedora, NVIDIA drivers, w/ KDE Plasma. Thanks!

happy to hear that, congrats