bambulab/BambuStudio #7346: Embedded text not slicing

andrewdavis360, 2025-06-20:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5

### Printer

Bambu Lab X1C

### How to reproduce

1. import model
2. add text element in bambu studio
3. set text to be 0 thickness (defaults to 0.01)
4. Set embedded text to 0.5 or any thickness
5. slice

### Actual results

on slicing text added to the model disapears

### Expected results

text should be sliced to be level with model face and embedded

### Project file & Debug log uploads

[Tile.3mf.zip](https://github.com/user-attachments/files/20826988/Tile.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-24:
Hello, can you download QuickRecorder software and record a GIF of the operation video and upload it? Thank you                         [lihaoyun6/QuickRecorder: A lightweight screen recorder based on ScreenCapture Kit for macOS / 基于 ScreenCapture Kit 的轻量化多功能 macOS 录屏工具](https://github.com/lihaoyun6/QuickRecorder/releases/tag/1.6.5)

zhimin-zeng-bambulab, 2025-06-26:
@andrewdavis360  When the faces of two parts overlap, the coloring takes priority. The bottom of your model is painted white, so it is displayed as white. I cleared all the coloring information, repainted, and saved the file. You can try it.

[Tile-save-as.zip](https://github.com/user-attachments/files/20927497/Tile-save-as.zip)