bambulab/BambuStudio #7356: 使用多屏幕分屏时软件在副屏打开会卡死的问题

Jerry0361, 2025-06-21:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 series

### How to reproduce

笔记本电脑接显示器，使用拓展屏幕模式，打开Bambu Studio，将软件拖进副屏(另一个屏幕)，关闭Bambu Studio，再次打开Bambu Studio。

### Actual results

Bambu Studio在打开过程中卡死

[image]

[image]

### Expected results

Bambu Studio应该被顺利打开。

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20846078/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

XinZhangBBL, 2025-06-23:
Halo. We are sorry for this bug, can you offer a log and dmp file for us, you can refer to https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue to get your log and dmp file.

XinZhangBBL, 2025-06-23:
[image]

XinZhangBBL, 2025-08-27:
Thank you for your feedback. Since necessary information has not been received for a long time, and this issue will be closed.
When you encounter any problems, feel free to provide feedback to us.

closed 2025-09-24 by XinZhangBambu