bambulab/BambuStudio #7364: 逐件打印时，如果某个零件是多色，其他零件是单色，没有出现擦拭塔（配置里是勾选了擦拭塔的）

g123413927, 2025-06-23:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows10

### Printer

Bambu H2D

### How to reproduce

普通打印，设置好几个逐件打印，其中有一个是多色。

### Actual results

导致多色零件打印失败（拉丝严重）

### Expected results

应该出现一个擦拭塔，保证多色打印不要串色，或者拉丝。

### Project file & Debug log uploads

[test.zip](https://github.com/user-attachments/files/20855863/test.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

hotellonely, 2025-06-23:
逐件打印目前会直接关掉擦料塔，所以这确实不是一个bug，是之前x1时代留下的“功能”。我也认为应该允许多擦料塔以及允许设置每个件开关擦料塔。这对于H2D之类的大盘机器会非常有用。可以显著降低人工成本。

g123413927, 2025-06-24:
好的，谢谢，不过确实有逐件打印有擦拭塔的需求。

closed 2025-06-24 by g123413927