bambulab/BambuStudio #7385: KSR FDM TEST WITH ERRORS IN NUMERATION

luis88luis, 2025-06-24:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

W11

### Additional system information

LAPTOP G15

### Printer

N/A

### How to reproduce

ADD PRIMITIVE
SELECT KSR FDM TEST 

### Actual results

THE TEXT HAS A WRONG NUMERATION IN THE CYLINDERS: 0.2 0.3 0.4 0.3 0.5 

### Expected results

IT SHOULD BE 0.2 0.3 0.4 0.5 0.6

This is so important because people is going to take this data and design and prepare their files with it. So maybe they are using a wrong value considering the wrong reading of their test result. 

### Project file & Debug log uploads

[asd.zip](https://github.com/user-attachments/files/20886771/asd.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

luis88luis, 2025-06-26:
Hi, I made a rework on the file and named it V5. Text are now corrected.

[testv5.zip](https://github.com/user-attachments/files/20930040/testv5.zip)

Haidiye00, 2025-06-27:
Hello, you're amazing. Let me help you inquire internally if it's possible to pass~

Haidiye00, 2025-11-30:
[image]
Hello, may I ask why it used to be a cylinder, but now it has turned into a frustum?

Haidiye00, 2026-01-30:
Hello, the new version has been released. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.05.00.65
If there's no issue, please help close it. Thank you