bambulab/BambuStudio #7393: Camera does nit work with latest update

lobert, 2025-06-25:
### Bambu Studio Version

1.08.02.00

### Where is the application from?

Bambu Lab Official website

### OS version

Sequoia 15.6 MACOS

### Additional system information

Mac Mini
Apple M2
16 GB memory

### Printer

Bambu P1P

### How to reproduce

Camera does not work when you press the play button

### Actual results

Camera does not work 

### Expected results

The camera should work

### Project file & Debug log uploads

[Open Closed.3mf.zip](https://github.com/user-attachments/files/20907578/Open.Closed.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lobert, 2025-06-27:
> BoBoYu-BamBu
>  left a comment
> (bambulab/BambuStudio#7393)
> @lobert <https://github.com/lobert> Thank you for your feedback. Could you please provide the log file and a screenshot of the playback failure?

It’s not a playback failure. The new software update shows a black screen when you use the camera. I’ve downgraded to the previous software version, and now everything works.

milk-pure, 2025-08-05:
Hello, could you please provide the studio's log files from when the issue occurred? We'll analyze them.