bambulab/BambuStudio #7395: Upgraded P1P to P1S shows up again as a P1P (Regression)

mbaran5, 2025-06-25:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab P1P w/ Official P1S Upgrade Kit

### How to reproduce

1. Ensure "Official Upgrade Kit" is selected on printer menu
2. Choose P1S and Plate type
3. Click "Sync Printer Information"
4. Receive error "The currently selected machine preset is inconsistent with the connected printer type. Are you sure to continue syncing?"

### Actual results

[image]

The machine is represented in the JSON as a P1P with model_id "C11"
```
"display_name": "Bambu Lab P1P",
"enable_set_nozzle_info": false,
"engineer": "00.06.03.51",
"ftp_folder": "sdcard/",
"has_cali_line": false,
"model_id": "C11",
"print": {
"ams": {
"ams": [
```
even though this line indicates it should assume the identity of C12 https://github.com/bambulab/BambuStudio/blob/a16cf773549d6731a80e6f87b476c450a0f13702/src/slic3r/GUI/SelectMachine.cpp#L1959

### Expected results

The machine should sync without error. 

### Project file & Debug log uploads

[debug_Wed_Jun_25_14_01_39_31680.log](https://github.com/user-attachments/files/20911566/debug_Wed_Jun_25_14_01_39_31680.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

mbaran5, 2025-06-25:
I also notice in the JSON that "printer_is_enclosed" is coming up false

```"printer_is_enclosed": false,```

mbaran5, 2025-07-01:
Please let me know if there is anything else I can share to help this issue. It seems similar, but not identical to:

https://github.com/bambulab/BambuStudio/issues/6208
https://github.com/bambulab/BambuStudio/issues/6204
https://github.com/bambulab/BambuStudio/issues/6149
https://github.com/bambulab/BambuStudio/issues/6181
https://github.com/bambulab/BambuStudio/issues/6148

zhimin-zeng-bambulab, 2025-07-08:
Thanks for reporting this issue, we will check it.

fahaaaxiki, 2025-07-21:
hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)

[image]