bambulab/BambuStudio #7405: Potential arbitrary code execution vulnerability in Bambu Studio

piuppi, 2025-06-26:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Studio

### How to reproduce

Hello Team.

In October 2024, I reported a vulnerability to the Bambu Lab Security Team via email at security(at)bambulab.com. The issue could potentially allow arbitrary code execution in the user context via Bambu Studio.

In January 2025, I received a brief reply apologizing for the delay, but since then I haven’t received any further updates, despite multiple follow-up emails requesting a status or acknowledgment.

I'm opening this issue here to raise awareness and encourage the Security Team to look into the matter. I believe it's important to maintain transparent and timely communication when it comes to security vulnerabilities that may pose real risks to users.

I’m available to provide further technical details in private or through any secure channel you prefer.

Thank you and Regards,
Gianluca Palma

### Actual results

 ...

### Expected results

... 

### Project file & Debug log uploads

 ...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2025-06-28:
Hello @piuppi 

Thanks for your suggestion

I have checked with our security team
and found the detailed information there

And we have planned to add the verify logic in the following 1-2 versions

piuppi, 2025-06-29:
Hello @lanewei120 

Thanks for the follow-up — I'm glad to hear that you’ve reviewed and analyzed the vulnerability report, and that a fix is being planned.

If there's anything I can do to assist with testing or validation, feel free to let me know.

I’d also kindly ask that you keep me updated through this issue and let me know once the patch has been officially released. I’d really appreciate it!

Looking forward to seeing the fix in an upcoming version.

Thanks and Regards,
Gianluca Palma

Ir1sk1dd0, 2025-09-01:
Hi @piuppi, that sounds interesting! But to exploit the issue, were you actually able to execute code, or is it just a potential risk?

piuppi, 2025-09-02:
> Hi [@piuppi](https://github.com/piuppi), that sounds interesting! But to exploit the issue, were you actually able to execute code, or is it just a potential risk?

Hi @Ir1sk1dd0 ,  yep, it’s exploitable. As you can see in the screenshot below, when bambu-studio.exe loads, it spawns my payload, so code execution is achieved. I initially wrote 'potential' in the issue title just to avoid giving too many public details :)

[image]

piuppi, 2025-10-08:
Hi @lanewei120, MITRE reserved CVE-2025-57521 for this vulnerability; you can use it as reference for tracking and remediation. Details will be published once the fix is out. thank you!

piuppi, 2025-10-15:
Hi @lanewei120, @walterwongbbl, I noticed that a new version of Bambu Studio has been released (v. 2.3.0.70); is there any update regarding the fix? I’m available for any recheck. Thanks!

piuppi, 2025-10-16:
> Hi [@piuppi](https://github.com/piuppi)
> 
> yes, this issue is fixed, please have a check on this commit
> 
> commit [9697631](https://github.com/bambulab/BambuStudio/commit/969763122f06e649350033eea95fbfe08d85bbbd) Author: haolin.tian [haolin.tian@bambulab.com](mailto:haolin.tian@bambulab.com) Date: Wed Sep 24 11:21:30 2025 +0800
> 
> ```
> FIX: verify cert when loading networking
> 
> this fixes issue of CVE-2025-57521 from MITRE
> 
> jira: STUDIO-13186
> Change-Id: I46df276a861f4d2bfec14839bc48355c63c57716
> ```

Hi @lanewei120, thanks for the info. I analyzed the commit and tested the new version of Bambu Studio (v2.3.0.70). I can confirm that the fix is effective and the network plugin can no longer be hijacked.
As far as I’m concerned, this vulnerability can be considered remediated.
Many thanks to the whole team for paying attention to the Bambu Lab product’s security.