bambulab/BambuStudio #7406: Does not remember the selected profile

gogsi1, 2025-06-26:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab H2D, Bambu lab X1 Carbon

### How to reproduce

1. Select my settings and plastic profile for Bambu Lab x1.
2. If you close the program or switch to another printer, the factory settings and plastic profiles will be selected, and you have to change them to your own each time.

### Actual results

It changes to factory profiles every time.

### Expected results

I want the selected settings profile and the list of plastics to remain the same as I selected them and not change to factory settings.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20929362/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-06-27:
@gogsi1 Sorry, I don't quite understand what you mean. Could you provide a screen recording or a screenshot to describe your problem in detail?

JohnSmithCYM, 2025-06-27:
Thanks for your feedback, the reason why you change filament preset and it changes due to the preset of filament or process is bind to the machine preset. So if there is a filament preset only bind to X1C, when you change to other machine preset, it will use a default filament. If you want to change the printer and the filament not change, you should add the filament filament settings of all printer preset and it will peform like this.

[image]

gogsi1, 2025-06-27:
@JohnSmithCYM, What about the profile settings? He's changing too.
If I select the printer's system profile, it works, but I choose my own profile. Now I will provide a video for understanding

gogsi1, 2025-06-27:
@JohnSmithCYM  When switching to another printer, the settings get lost. 
https://www.youtube.com/watch?v=s5pHoyniVrE

gogsi1, 2025-06-27:
@JohnSmithCYM  I have only one question, how do I make sure that the selected settings are saved, if this cannot be done, will they be fixed in the new version? It's tedious to go back to the settings every time.

tvixen, 2025-07-01:
> [@JohnSmithCYM](https://github.com/JohnSmithCYM) I have only one question, how do I make sure that the selected settings are saved, if this cannot be done, will they be fixed in the new version? It's tedious to go back to the settings every time.

The same goes for the color scheme... its not saved to the profile when exit, like it did before this version.