bambulab/BambuStudio #7422: Request support for macOS 26

xdfnet, 2025-06-28:
### Bambu Studio Version

2.1.1 Public Release

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 26

### Printer

P1S

### How to reproduce

Run App

### Actual results

Open the APP and report an error, unable to enter the program

### Expected results

Open the APP and report an error, unable to enter the program

### Project file & Debug log uploads

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------
Process:             BambuStudio [46592]
Path:                /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:          com.bambulab.bambu-studio
Version:             02.01.01.52 ()
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           com.bambulab.bambu-studio [9832]
User ID:             501

Date/Time:           2025-06-28 18:00:04.0406 +0800
Launch Time:         2025-06-28 18:00:02.0498 +0800
Hardware Model:      Mac16,1
OS Version:          macOS 26.0 (25A5295e)
Release Type:        User

Crash Reporter Key:  EB44EE13-345C-E9C6-C0D6-2A1F58368009
Incident Identifier: B8A59ED9-81B0-497D-BF76-9BA6C16F77E5

Sleep/Wake UUID:       1F0B3F27-198A-4FCF-8446-8A310CBAFC34

Time Awake Since Boot: 66000 seconds
Time Since Wake:       12740 seconds

System Integrity Protection: enabled

Triggered by Thread: 0  bambustu_main

Exception Type:    EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:   0x0000000000000001, 0x0000000000000000

Termination Reason:  Namespace SIGNAL, Code 11, Segmentation fault: 11
Terminating Process: exc handler [46592]

VM Region Info: 0 is not in any region.  Bytes before following region: 4371529728
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
[... 512 lines cut ...]
  ],
  "experiments" : [

  ]
}
}

Model: Mac16,1, BootROM 13822.0.166.0.1, proc 10:4:6 processors, 16 GB, SMC 
Graphics: Apple M4, Apple M4, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Display: Redmi 27 NU, 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition), MirrorOff, Online
Memory Module: LPDDR5, Micron
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4388), wl0: Jun 11 2025 15:18:24 version 23.10.965.2.41.51.188 FWID 01-d1048a97
IO80211_driverkit-1525.73 "IO80211_driverkit-1525.73" Jun 13 2025 20:18:17
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

### Checklist of files to include

- [x] Log file
- [ ] Project file

lanewei120, 2025-07-01:
@xdfnet 
also how about the previous version?
is it running well on your laptop

xdfnet, 2025-07-01:
> 您好，您能向我们提供购买笔记本电脑的链接吗？很抱歉，我们手头还没有这台笔记本电脑或其匹配的环境

Apple/苹果AI笔记本/2024MacBookPro14英寸 M4(10+10核) 16G 512G深空黑色纳米玻璃笔记本电脑

frank-pian, 2025-07-15:
Related to https://github.com/bambulab/BambuStudio/issues/7424