bambulab/BambuStudio #7424: Crashes on macOS26 Tahoe

xero-xer0, 2025-06-28:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 26 Tahoe

### Additional system information

CPU : Apple M2 Pro
RAM : 16GB

### Printer

Bambu Lab A1

### How to reproduce

1. Go to Preview tab
2. Click any box where you can enter numbers or letters.
3. Bambu Studio IMMEDIATELY Crashes.

### Actual results

I can't put any values into textbox. If I try to enter something in the box, it immediately crashes.

### Expected results

I should be able to put the value in the input box.

### Project file & Debug log uploads

[Logs.zip](https://github.com/user-attachments/files/20961307/Logs.zip)

[화면 기록 2025-06-28 오후 11.43.00.zip](https://github.com/user-attachments/files/20961308/2025-06-28.11.43.00.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

frank-pian, 2025-07-08:
@xero-xer0 Can you provide the macOS crash report ?

xero-xer0, 2025-07-08:
Of course. 

[bambustudio_crashreport.txt](https://github.com/user-attachments/files/21121024/bambustudio_crashreport.txt)

frank-pian, 2025-07-08:
> Of course. [@frank-pian](https://github.com/frank-pian)
> 
> [bambustudio_crashreport.txt](https://github.com/user-attachments/files/21121024/bambustudio_crashreport.txt)

This seems to be an issue between macOS and wxWidgets. I’ll try to look for a solution on the wxWidgets side.

frank-pian, 2025-07-15:
@xero-xer0 
I tested the new macOS 26 Beta (25A5306g), and it runs fine. You can try updating your system and deleting the configuration files under /Users/USER_NAME/Library/Application Support/BambuStudio/BambuStudio.conf

[image]

frank-pian, 2025-07-15:
> [@frank-pian](https://github.com/frank-pian) I tried to reinstall, and I think I finally found the reason. If the language setting is English, it works fine. But after I set it to Korean and press the number field, it immediately crashes. Thank you for your help, and I hope it will be fixed!

I tried switching to Korean, and it didn't crash.
[image]

xero-xer0, 2025-07-15:
@frank-pian It does not crash when just running the program. 
It crashes when you are trying to type something in the "preview" section. like trying to change "Layer Height" values, and other number or text values.

xero-xer0, 2025-07-15:
https://github.com/user-attachments/assets/e20a4e75-3f4f-4828-b589-e5e42516ea6d

xero-xer0, 2025-07-15:
That's weird. Of course I cleaned the directories. and fully reinstalled. 
I deleted the Bambu studio using AppCleaner, the utility that erases all related files, and I also double-checked the directory you've told me. 
And I re-downloaded from Bambu studio official download site and installed it again. but The same issue appeared.

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)