bambulab/BambuStudio #7426: Copy to clipboard measurements copies useless text together with the number

vanowm, 2025-06-28:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 x64

### Printer

X1C

### How to reproduce

1. Using measuring tool measure a length or a distance
2. Click `Copy` button next to the number

[image]

### Actual results

It copies `Distance: 180.115 mm`

### Expected results

`180.115`

Users use the tool to calculate, not to post on social media.

Every single number in that window should be accessible to be copied to clipboard as well.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vanowm, 2025-06-30:
Like to calculate how much I need to move a part. You copy a number and paste into a calculator...

vanowm, 2025-07-07:
I don't see anything fixed in this screenshot... if it copies anything other than the number - it's not fixed.

Haidiye00, 2025-07-07:
hello,The new fix only considers numbers and their units. You can see the green box content in the screenshot

Haidiye00, 2025-07-07:
[image]
Hello,you are right, that has been changed to this. Thank you for your feedback~

Haidiye00, 2025-07-07:
Sorry, currently copying from other places is not supported

vanowm, 2025-07-27:
Thank you it's working fine now

closed 2025-07-27 by vanowm