bambulab/BambuStudio #7430: Bambu Studio keeps freezing on Ubuntu. Happens on every 2nd day.

danergo, 2025-06-29:
### Bambu Studio Version

1.10.2.64

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 24.04.2 LTS

### Additional system information

Here is your gdb log:
```

(gdb) bt
#0  0x00007ef3eb11ba9a in __GI___libc_read (nbytes=135168, buf=0x7ef3eb32d010, fd=5) at ../sysdeps/unix/sysv/linux/read.c:26
#1  __GI___libc_read (fd=5, buf=0x7ef3eb32d010, nbytes=135168) at ../sysdeps/unix/sysv/linux/read.c:24
#2  0x00007ef3eb3639fa in ?? () from /lib/x86_64-linux-gnu/libfuse.so.2
#3  0x00007ef3eb3654bc in ?? () from /lib/x86_64-linux-gnu/libfuse.so.2
#4  0x00007ef3eb363bde in fuse_session_loop () from /lib/x86_64-linux-gnu/libfuse.so.2
#5  0x000000000040e409 in fusefs_main ()
#6  0x0000000000404dbb in main ()

```

This application is behaving very ugly on Ubuntu. It keeps freezing, but a strange way: in case I click on the menu, it updates the UI then freeze again. Every single click on the menu causes 1 exact render on the UI.

=> Unusable.

What do you recommend?

### Printer

X1C

### How to reproduce

Just use it on Ubuntu.

### Actual results

Constant freeze.

### Expected results

No freeze.

### Project file & Debug log uploads

Attached the debug above.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2025-07-29:
Has the default desktop environment of Ubuntu been changed to KDE?

danergo, 2025-07-29:
No, everything is left on default.

JohnSmithCYM, 2025-08-14:
Can you reproduce this problem with the lastest 2.2.0.85?

EmRodDev, 2025-09-16:
Same on Fedora Workstation 42