bambulab/BambuStudio #7431: Preview page rendering text blurred

luciotorelli, 2025-06-29:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Bazzite 42 (FROM Fedora Silverblue) - Gnome

### Additional system information

Framework Laptop 13 _AMD Ryzen 7040Series_
AMD Ryzen™ 5 7640U w/ Radeon™ 760M Graphics × 12

### Printer

P1S

### How to reproduce

Download either AppImage or Flatpak > Launch > Preview.

### Actual results

[image]

### Expected results

Rendered Text

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [x] Log file
- [x] Project file

maximushugus, 2025-06-29:
I have the same problem on an PC with Intel UHD 620 running Arch Linux with KDE on Wayland.
BambuStudio comes from [AUR](https://aur.archlinux.org/packages/bambustudio-bin)

MackBambu, 2025-07-28:
The issue is very similar to an old ImGui bug. Have you changed the display scaling? You can try setting it to 100% to see if it helps.