bambulab/BambuStudio #7432: Crashing every time my Macbook sleeps since updating to 2.1.1.52

Hyp3rDrive, 2025-06-29:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.4.1

### Additional system information

M1 Macbook Air

### Printer

Bambu Lab X1c

### How to reproduce

Bambu studio open on the device screen when the MacBook goes to sleep. Bambu studio 2.1.1.52 MacOS Sonoma 14.4.1

### Actual results

On waking the Macbook Bambu Studio is unresponsive and needs to be force quit.

### Expected results

Bambu Studio should not be frozen upon waking the MacBook from sleep.

### Project file & Debug log uploads

[Crashes and 3mf.zip](https://github.com/user-attachments/files/20967494/Crashes.and.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Scroeder, 2025-06-30:
Can confirm that... Thought the updated already fixed it but unfortunately it did not.

XinZhangBBL, 2025-07-01:
@newoldschool @Hyp3rDrive @Scroeder @hotellonely Sorry for this bug. Please attach the text of MAC crash report.

Scroeder, 2025-07-06:
> [@newoldschool](https://github.com/newoldschool) [@Hyp3rDrive](https://github.com/Hyp3rDrive) [@Scroeder](https://github.com/Scroeder) [@hotellonely](https://github.com/hotellonely) Sorry for this bug. Please attach the text of MAC crash report.

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [59351]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               02.01.01.52 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-07-05 23:19:16.1582 +0200
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        356A78A2-3E03-1A6C-8BEB-AFE2C13B11BC

Sleep/Wake UUID:       6B648CAC-171F-4C2A-B9C6-08502E37255D

Time Awake Since Boot: 1100000 seconds
Time Since Wake:       3 seconds

System Integrity Protection: enabled

Crashed Thread:        26

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [59351]

VM Region Info: 0 is not in any region.  Bytes before following region: 4339040256
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102a08000-106bec000    [ 65.9M] r-x/r-x SM=COW  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio

Thread 0:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x1030db314 nlohmann::detail::output_string_adapter<char, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>::write_characters(char const*, unsigned long) + 0
1   BambuStudio                   	       0x102b70a90 nlohmann::detail::serializer<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>>::dump(nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>> const&, bool, bool, unsigned int, unsigned int) + 2200
2   BambuStudio                   	       0x102b70ac4 nlohmann::detail::serializer<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>>::dump(nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>> const&, bool, bool, unsigned int, unsigned int) + 2252
3   BambuStudio                   	       0x102b70ac4 nlohmann::detail::serializer<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>>::dump(nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>> const&, bool, bool, unsigned int, unsigned int) + 2252
4   BambuStudio                   	       0x1030db190 nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>>::dump(int, char, bool, nlohmann::detail::error_handler_t) const + 556
5   BambuStudio                   	       0x103d85704 Slic3r::MachineObject::parse_json(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool) + 82080
6   BambuStudio                   	       0x103919668 wxAsyncMethodCallEventFunctor<Slic3r::GUI::GUI_App::init_networking_callbacks()::$_20::operator()(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) const::'lambda'()>::Execute() + 660
7   BambuStudio                   	       0x105efd7d0 wxEvtHandler::ProcessEventLocally(wxEvent&) + 172
[... cut ...]
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Dec  7 2024 04:41:25 version 18.90.5.0.7.8.177 FWID 01-02f4dc53
IO80211_driverkit-1475.39 "IO80211_driverkit-1475.39" Apr 18 2025 20:10:40
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)

closed 2025-12-01 by Hyp3rDrive