bambulab/BambuStudio #7437: Prime Tower faulty path generation, causing nozzle bumps on tower 擦料塔错误生成路径，导致喷嘴与塔撞击

hotellonely, 2025-06-30:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Any

### Printer

Bambulab H2D

### How to reproduce

1. Download 3mf file.
2. Slice as is. Or use any default profiles, but leave support enabled and also keep using the support interface filament.

[image]

3. **Slice and go to preview page, have a detailed look at the prime tower on layer 455.**

[image]

In fact, there are many more layers with the same issue. The white line goes over already printed parts, causing the nozzle to hit hard on the prime tower, introducing print failures.

### Actual results

The generated prime tower creates faulty tool path that would cause the nozzle to bump the prime tower and eventually knock the prime tower off the plate.

[image]

I literally have no idea how it's generated to be like this.

### Expected results

Expected result should be a prime tower with clean and non-self crossing tool paths, like this:
[image]

### Project file & Debug log uploads

the 3mf file is too large to be uploaded to github, thus I uploaded it to my google drive.

https://drive.google.com/file/d/1q-H-6bZnljjPk-pBNc474RJUWtTyqPPu/view?usp=drive_link

### Checklist of files to include

- [ ] Log file
- [x] Project file

hotellonely, 2025-06-30:
This issue was initially brought up by Bambulab Forum User @podje, thanks to his IPCAM recording and 3mf file I was able to pin this issue out. The original discussion was here: 
https://forum.bambulab.com/t/prime-tower-knocked-down-this-is-not-an-adhesion-problem/172727/45
You can also find sliced gcode, IPCAM recording and other things there.

hotellonely, 2025-07-03:
any update on this..?

zhimin-zeng-bambulab, 2025-07-03:
Cannot reproduce this issue. Could you provide steps to reproduce it?

hotellonely, 2025-07-03:
> Cannot reproduce this issue. Could you provide steps to reproduce it?

https://www.youtube.com/watch?v=YH3ZHk3Xvnc

I recorded the screen and showed how to reproduce it in a minute, I don't understand how you couldn't reproduce it though

podje, 2025-07-13:
Any update on this one, as the issue has occurred multiple times across multiple prints for me and clear steps have been given to reproduce.

zhimin-zeng-bambulab, 2025-07-14:
Please wait patiently for a while, we will release a version soon.

hotellonely, 2025-07-14:
Zhimin my legend of all stars

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)