bambulab/BambuStudio #7448: Clearing the search field in Online Models tab does not clear the search (No refresh option)

ThinkingParadox, 2025-07-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

Go to the Home section of Bambu Studio. 
Go to Online Models
Search for anything
Click the X button to clear the search. 

### Actual results

It will not clear the search and refresh 

### Expected results

It should refresh and go back to the default view of Maker World. Please if possible add a refresh button to this tab. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-07-02:
@ThinkingParadox  Thank you for your feedback. Please submit a work order on the mall website based on the screenshot provided. The mall team will follow up on this issue afterwards.this page is embedded within the studio software in the mall. Therefore, all issues related to this page will be handled by the mall side. I have already informed the mall team and they will make the necessary changes. Thank you.
[image]

BoBoYu-BamBu, 2025-07-02:
@ThinkingParadox Oh, sorry. Please use this link to open it:https://makerworld.com/en/3d-models

ThinkingParadox, 2025-07-02:
Thank you! I have opened a support ticket!

closed 2025-07-02 by ThinkingParadox