bambulab/BambuStudio #7449: Crash when sending plate to H2D printer

matt-e, 2025-07-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux (Linux 6.15.4-arch2-1)

### Additional system information

CPU: AMD Ryzen 9 9950X (32) @ 5.76 GHz
GPU: NVIDIA GeForce RTX 4090
Memory: 96GB

### Printer

H2D

### How to reproduce

1. Slice a print
2. In the preview view, click the dropdown arrow next to print and choose "Send"
3. Click send

### Actual results

The program shows a dialog with the message "Try to connect" - and then Bambu studio crashes

### Expected results

The sliced project should be sent to the printer.

### Project file & Debug log uploads

gdb backtrace: 

```
Thread 80 "bambustu_main" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fb44edfe6c0 (LWP 82796)]
0x00007fb702ea774c in ?? () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007fb702ea774c in ?? () from /usr/lib/libc.so.6
#1  0x00007fb702e4ddc0 in raise () from /usr/lib/libc.so.6
#2  0x00007fb702e3557a in abort () from /usr/lib/libc.so.6
#3  0x00007fb702e365c9 in ?? () from /usr/lib/libc.so.6
#4  0x00007fb702f34a09 in __fortify_fail () from /usr/lib/libc.so.6
#5  0x00007fb702f34374 in __chk_fail () from /usr/lib/libc.so.6
#6  0x00007fb702f35c37 in __snprintf_chk () from /usr/lib/libc.so.6
#7  0x00007fb6d37518bd in BambuTunnelLocal::start(int) () from /home/matt/.config/BambuStudio/plugins/libBambuSource.so
#8  0x00007fb6d373f4d7 in BambuTunnel::start_stream_ex(int) () from /home/matt/.config/BambuStudio/plugins/libBambuSource.so
#9  0x0000000001e0ac7c in PrinterFileSystem::Reconnect(boost::unique_lock<boost::mutex>&, int) ()
#10 0x0000000001e1abd7 in PrinterFileSystem::RecvMessageThread() ()
#11 0x0000000001e1bfeb in boost::detail::thread_data<PrinterFileSystem::Attached()::{lambda()#1}>::run() ()
#12 0x000000000238c917 in thread_proxy ()
#13 0x00007fb702ea57eb in ?? () from /usr/lib/libc.so.6
#14 0x00007fb702f2918c in ?? () from /usr/lib/libc.so.6
```

### Checklist of files to include

- [x] Log file
- [x] Project file

XedrikJr, 2025-07-07:
Having same issue with my X1 Carbon. I only started experiencing this issue after updating the printer firmware to this latest version. 

Bambu Studio 2.1.1.52
Kubuntu 24.04
Intel Core i5, 16GB RAM
Bambu X1 Carbon printer firmware 01.09.01.00
AMS-1 version: 00.00.06.75

After slicing, I click the down-arrow on the "Print Plate" button to select "Send", and on clicking "Send", Bambu Studio very briefly pops up the expected dialog (showing the printer name, giving an opportunity to name the file, confirming Send) but immediately crashes.

Using the same project file, same slicer settings, same everything, if instead of using "Send", I hit "Print Plate" it sends the job just fine, but involves cloud nonsense instead of just sending to the printer.  Network connectivity to the printer is otherwise as expected. I can view the device information, update the firmware, watch the camera stream, sync filaments.

[Project Crashes On Send.zip](https://github.com/user-attachments/files/21111171/Project.Crashes.On.Send.zip)

[log.zip](https://github.com/user-attachments/files/21111174/log.zip)

milk-pure, 2025-09-29:
Please update to the latest software and test again.

XedrikJr, 2025-11-08:
> Can you reproduce this problem with [lastest version](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85)?

Issue appears to be fixed in latest version.

BoBoYu-BamBu, 2025-12-02:
Thank you for your feedback. This issue has been fixed. Please update to the latest version of Studio for verification. Download link: https://bambulab.cn/zh-cn/download/studio

closed 2025-12-02 by BoBoYu-BamBu