bambulab/BambuStudio #7452: Avoid crossing wall not working for first travel move in layer unless retracted

fhufi, 2025-07-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro

### Printer

Bambu Lab X1E

### How to reproduce

- New project, Standard Process
- Object with concave regions or holes
- Slice -> many travel moves across concave walls -> scarring observed in many prints

[image]

- Check option "Avoid crossing wall"
- Slice again

### Actual results

[image]

Wall crossings are reduced, but many remain -> still scarring of concave surfaces
Gcode shows a straight travel move at the beginning of each layer during z-hop

[image]
[image]

Workaround: in Filament Setting Overrides, check box Retract when change layer but DO NOT check the box on the right  (description of second box unclear) -> Result: Travel move now avoids wall crossing!

[image]
[image]

### Expected results

Expected behavior is that **all** possible wall crossings should be avoided when option is checked, also during z-hop, without the need for individual Filament Setting Overrides. 

In case the behavior of the option is "as intended", I strongly disagree and kindly ask for reconsideration. For many materials (try TPU!) z-hop alone is not sufficient to eliminate scars and stringing. 

### Project file & Debug log uploads

[Avoid_crossing_wall.zip](https://github.com/user-attachments/files/21020396/Avoid_crossing_wall.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

fhufi, 2025-07-03:
Related issue #7240 on MacOS version

sakomoto987, 2025-07-21:
Hi everyone,

Just wanted to share some findings regarding the Avoid Crossing Wall feature in Bambu Studio, in case others are running into similar issues.

I’ve tested the recent 2.2.0 Public Beta, and while there are some improvements — notably, the nozzle no longer travels outside the model on curved areas — the core issue still persists: travel moves are still crossing internal walls unnecessarily, even when Avoid Crossing Wall is enabled.

You can see this clearly in the screenshots attached. These crossings occur in situations where the path planner could clearly avoid them. What makes this especially problematic is when printing flexible materials like TPU, where clean travel paths really matter for quality as mentioned.

One additional detail: I’m using a custom seam setting, but the exact same configuration works perfectly in OrcaSlicer, and was previously handled correctly in older versions of Bambu Studio.

To summarize:
	•	Bambu Studio 2.2.0 beta improves outer travel behavior on curves ✅
	•	Avoid Crossing Wall still doesn’t prevent unnecessary internal wall crossings ❌
	•	Custom seam settings may play a role, but it’s not an issue in OrcaSlicer or earlier Bambu Studio releases

Happy to share more info if needed — hoping this helps others or leads to further improvements!

[image]
[image]

Noisyfox, 2025-07-23:
https://github.com/SoftFever/OrcaSlicer/pull/10185#issuecomment-3108095779