bambulab/BambuStudio #7454: Sidebar  doubles in width, impossible to revert

hirvikoski, 2025-07-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

Any printer

### How to reproduce

After having opened Bambu Studio via remote desktop the scaling that occurs permanently screw up the width of the sidebar

### Actual results

[image]

### Expected results

Fix the scaling issues and make the side bar resizable

### Project file & Debug log uploads

no

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-07-03:
https://github.com/Haidiye00/BambuStudio/actions/runs/15534911769
[image]
This is a temporary version, please try it out~thanks

1.open this zip software,come to preivew tab, then close software. find this BambuStudio.conf,and modify enable_sidebar_resizable 's value
[image]
2.reopen this  zip software, come to preivew tab,then press "ctrl+w"
[image]

then you can adjust sidebar's width

Hello, in the next version, we will default to supporting adjustable sidebars~

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~

Hello, please note that after opening the software, it is recommended that you press Ctrl+W before adjusting the width