bambulab/BambuStudio #7455: No Minimum Suggested Flush Volume

Malice890, 2025-07-02:
### Bambu Studio Version

2.1.1.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 / 11

### Printer

All

### How to reproduce

Open flushing volumes menu

### Actual results

Suggestion line shows 0 as the minimum, and there is no warning if a volume is set very low (under 50).

[image]

### Expected results

Minimum flushing volume is above 0 (normally 50-100), and gives a warning when setting a volume below that.

[image]

### Project file & Debug log uploads

[petg supports pla.zip](https://github.com/user-attachments/files/21026009/petg.supports.pla.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)

Malice890, 2025-07-23:
> Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)

I will check it out in the Public Release. 
Thank you.

Malice890, 2025-08-13:
Seems to be working as it should now, thank you.

closed 2025-08-13 by Malice890